pepe2k/u-boot_mod #48: ar9331 u-boot RAM version

ander-wrt, 2015-01-26:
I've tried to use RAM version of u-boot for router debrick,
but there is very strange situation.

RAM version can't start stable for each launch.
Let me describe in details:

I use mr3020 v1.8 router with ar9331 on board.
I use j-link in conjunction with openocd 0.8.0.
I use pll and RAM init sequence from here:
https://wikidevi.com/wiki/TP-LINK_TL-MR3020#JTAG
Commands for openocd are:

halt
reset init
load_image uboot_for_tp-link_tl-mr3020__RAM.bin 0x80100000
resume 0x80100000

Image loaded successfully. I can verify it.
I can see RAM version logo for each start,
but usually it hangs at relocate_code function
near code

1:
        lw         t3, 0(t0)
        sw        t3, 0(t1)
        addu    t0, 4
        ble       t0, t2, 1b

I can halt processor,but RAM is not accessible anymore.
I've tried different version of u-boot from git,
but behaviour is the same.

It can start once for 20...30 attempts approximately.

JTAG frequency change doesn't help.
CPU/AHB frequency change doesn't help.
Router works well if I desolder flash and programm it.
But I would like to use RAM version of u-boot.

Where is my mistake?

pepe2k, 2015-01-28:
Hello @ander-wrt,

I hope to have some time this weekend to take a look on this problem.
Please, just let me know what AR9331 revision does have your MR3020.

Regards

ander-wrt, 2015-01-28:
Hi pepe2k,

I have different revisions: AR9331-AL1A and AR9331-AL3A.
Revision does not matter. bootloader hangs on both at the same point.
RAM version never reach any hardware breakpoint behind code I mentioned above.
It looks like something wrong with RAM controller initialization.
I can go step-by-step through relocate_code function
but u-boot hangs later on any intensive RAM operation if I resume.
Intensive RAM operation I mean memory move, copy and so on.

Regards

ander-wrt, 2015-02-05:
Hi pepe2k,

Have you had a chance to take a look on this problem or not yet?

Regards

pepe2k, 2015-02-15:
Hello @ander-wrt,

Sorry, I'm busy with other projects now.

Regards

closed 2016-06-15 by pepe2k

manfeel, 2019-05-18:
> I have different revisions: AR9331-AL1A and AR9331-AL3A.
> Revision does not matter. bootloader hangs on both at the same point.
> RAM version never reach any hardware breakpoint behind code I mentioned [...]

Hello @ander-wrt,

YES, my friend! I have encountered exactly the SAME problem! Intensive RAM operation(e.g. in function tinf_crc32 and relocate code and ...) will cause ram code crash, and subsequent memory read and write operations through openocd will return an error. I'am using JLink v8 + openocd 0.10. I guess, will it be caused by the cache settings? 

Regards