pepe2k/u-boot_mod #50: Overclocking AR9331 router

efidrin, 2015-02-26:
Dear pepe2k , need some help.

I have TP-Link WR741ND v4.2 , witth 16mb FLASH, and 64mb RAM.
Using your wonderfull moded U-Boot, its works perfect.

Then I decide change quartz , from 25MHz to 40MHz.

I find that signal on GPIO0 at start, must be signal 1 (instad 0 for 25MHz).
First , I replace quartz.
Then I cut the WLAN led, and connect "+" with 3,3V .
[image]
At the last, I rebuild U-Boot adding to  wr740n_v4_config
line 
	@echo "#define CONFIG_40MHZ_XTAL_SUPPORT            1" >> include/config.h

After turning on device, on console I have seen some progress(like booting) , but with
unknown symbols.
I have tried different speed of serial port? but nothing.

P.S. When i change 25MHz quart on 33MHz, i saw boot log whis 115200/25*33=152064 speed
using original boot image

pepe2k, 2015-03-15:
Hello @efidrin,

Sorry for late reply.
Take a look at this branch: https://github.com/pepe2k/u-boot_mod/tree/ar9331_new_uart_driver, there is new UART driver.

Regards

efidrin, 2015-03-16:
I treid , but if I add CONFIG_40MHZ_XTAL_SUPPORT option
getting "Image too big " message
can I set the default boot size 128 kb

pepe2k, 2015-03-16:
Remove some commands which you don't need:
https://github.com/pepe2k/u-boot_mod/blob/master/u-boot/include/configs/ap121.h#L789

closed 2015-04-27 by pepe2k