pepe2k/u-boot_mod #52: support flash SST25VF032B and SST25VF064C

TolyanPoltava, 2015-03-13:
pepe2k, please help.
*********************************************
*   U-Boot 1.1.4  (Mar 12 2015, 21:03:30)   *
*********************************************

AP123 (AR9341) U-Boot for TL-WR841N/D v8

DRAM:   64 MB DDR 32-bit
FLASH:  Unknown type (using only 4 MB)

Please, send request to add support
for your FLASH - JEDEC ID: 0xBF254A

CLOCKS: 535/400/200/25 MHz (CPU/RAM/AHB/SPI)

## Error: MAC is invalid, using fixed!

LED on during eth initialization...

## Error: MAC address in FLASH is invalid, using fixed!
## Error: MAC address in FLASH is invalid, using fixed!
Hit any key to stop autobooting:  0

uboot> erase 0x9f000000 +0x400000
Erase FLASH from 0x9F000000 to 0x9F3FFFFF in bank #1
Erasing: #######################################
         #########################

Erased sectors: 64

uboot> res

*********************************************
*   U-Boot 1.1.4  (Mar 12 2015, 21:03:30)   *
*********************************************

AP123 (AR9341) U-Boot for TL-WR841N/D v8

DRAM:   64 MB DDR 32-bit
FLASH:  Unknown type (using only 4 MB)

Please, send request to add support
for your FLASH - JEDEC ID: 0xBF254A

CLOCKS: 535/400/200/25 MHz (CPU/RAM/AHB/SPI)
Record \ erasing does not work.
http://lists.denx.de/pipermail/u-boot/2014-October/192519.html
added ar7240_flash.c
case 0xBF254A:
			flash_set_geom(SIZE_INBYTES_4MBYTES, 1024, SIZE_INBYTES_4KBYTES);
			puts("SST25VF032B (4 MB)");
			break;
log:
uboot> erase 0x9f000000 +0x400000
Erase FLASH from 0x9F000000 to 0x9F3FFFFF in bank #1
Erasing: #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         #######################################
         ##########

Erased sectors: 1024

uboot> res

*********************************************
*   U-Boot 1.1.4  (Mar 12 2015, 21:13:39)   *
*********************************************

AP123 (AR9341) U-Boot for TL-WR841N/D v8
...
In both cases, the erasure happens too fast
Please help me.
Thank you

pepe2k, 2015-03-15:
Hello @TolyanPoltava,

I don't have access to SST25VF032B/SST25VF064C FLASH chips.

You should add JEDEC IDs here (with different sector size): https://github.com/pepe2k/u-boot_mod/blob/master/u-boot/board/ar7240/common/ar7240_flash.c#L120

You may also need to change command for sector erase:
https://github.com/pepe2k/u-boot_mod/blob/master/u-boot/board/ar7240/common/ar7240_flash.h#L21

Regards

closed 2015-03-15 by pepe2k

bigotini, 2016-04-25:
Hi, I realize this is a closed issue, but I have the same problem.
I ordered a whole bunch of this SST25VF032 by mistake and now I can't use them.
I just need to know what files need editing so that I can compile, as I can't find the above files in the current git download.
Thanks

pepe2k, 2016-04-25:
Hello @bigotini,

Current version of my code supports **SFDP** (more here: https://github.com/pepe2k/u-boot_mod#supported-flash-chips) and as far as I know, those chips should support it too.

Have you tried to compile and run current version of the code?

Cheers,
Piotr

bigotini, 2016-04-25:
Hi Piotr thank you so much for taking the time to answer.
Yes, I edited the file "/u-boot/common/flash.c" which is where I found all the supported flash device definitions but it finishes with error 2. Here is the actual error:
"
flash.c: In function 'flash_manuf_name':
flash.c:76:7: error: 'FLASH_VENDOR_JEDEC_SST' undeclared (first use in this function)
flash.c:76:7: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [flash.o] Error 1
make[2]: Leaving directory `/home/su/u-boot_mod/u-boot/common'
make[1]: *** [common/libcommon.a] Error 2
make[1]: Leaving directory `/home/su/u-boot_mod/u-boot'
make: *** [tplink_wr740n_v4] Error 2"
I hope you can replicate, an than you for your great work
Luis

pepe2k, 2016-04-25:
Hello @bigotini,

I didn't ask you to edit the code. Don't touch it and try default version. If SFDP is supported on your FLASH chip/s, it should work (all needed parameters will be recognized automatically).

Cheers,
Piotr

bigotini, 2016-04-25:
Well I did that before my first post, but anyway I tried it without editing and it does finish compiling.
but when I write the u-boot.bin to the chip using programmer it does not boot. 
Do you think I should be getting something even if the flash is not supported?

pepe2k, 2016-04-25:
> Do you think I should be getting something even if the flash is not supported?

Yes, in most cases you should... proper FLASH chip support is needed only for erase/write.
What kind of board do you have?

bigotini, 2016-04-25:
I'm using WR-720N it only comes with 2M flash, an 16M ram.
Maybe the problem is that this chip is not SFDP compliant, this is the exact part #.
SST25VF032B-66-4I-S2AF I just checked the pdf and makes no reference to SFDP whatsoever.
But still I read on other forums that it is supported by u-boot on many other platforms.

pepe2k, 2016-04-25:
> Maybe the problem is that this chip is not SFDP compliant, this is the exact part #.

No, it doesn't have anything to do with booting.
SOC is able to boot from any compliant SPI NOR FLASH as it's a standard.

I think I know what is the real reason of your problem... WR720N has wrong bootstrap configuration and I forgot to setup static DRAM type for this board after last changes.

Please, add:

``@echo "#define CONFIG_BOARD_DRAM_TYPE_DDR1 1" >> include/config.h``

Somewhere here: https://github.com/pepe2k/u-boot_mod/blob/master/u-boot/Makefile#L364 and rebuild the code.

Let me know if this works for you.

bigotini, 2016-04-25:
I also have this other board based on the same chipset by Alfa Network:
https://wiki.openwrt.org/toh/alfa.network/hornet-ub
This one has 32M DDR , and boots with any tplink stock uboot and also your moded u-boot.
But when I try with the SST chip nothing boots.
Do you think I may have defective chips I already tried  2 out of 10 I have ordered.
Since I have tried 2 boards with same result do you think is worth trying the above fix in the code?

pepe2k, 2016-04-25:
Maybe you have problem somewhere else. How do you program that FLASH chip? Are you using verify functionality in your FLASH programmer?

Winbond W25Q128 is cheap, reliable and fully supported (U-Boot, Linux kernel), maybe buy them instead of that SST if they are so problematic.

bigotini, 2016-04-25:
I use home build programmer with SPI tinytools software from tele-sat.ru.
Use it all the time with all kinds of chips from intel, Mx,ST and others and this programmer recognizes erases and programs the SST chips just fine but somehow they don't boot.
Anyway I appreciate you help and keep up the good work.
One more question. Just out of curiosity since I know nothing about programming, should not SST CHIP information be declared like the rest of supported chips on that flash.c file.

pepe2k, 2016-04-25:
> Use it all the time with all kinds of chips from intel, Mx,ST and others and this programmer recognizes erases and programs the SST chips just fine but somehow they don't boot.

As last check, you can try image from last stable release (https://github.com/pepe2k/u-boot_mod/releases). As you can see at the begin of this issue, WR841Nv8 with SST25VF032B was working.

> One more question. Just out of curiosity since I know nothing about programming, should not SST CHIP information be declared like the rest of supported chips on that flash.c file.

Yes, when they are tested and confirmed to work.