pepe2k/u-boot_mod #54: TL-WR841ND Support for V9 with QCA9533-AL3A SoC?

Rolle77, 2015-03-17:
Hi everyone, ist there support in the modded bootloader für the TL-WR841ND V9, with the QCA9533-AL3A SoC (Manufacturer: Qualcomm-Atheros)? I have 3 pieces of this router and want to replace the flashchip with a 16Mb one... 

Greetings
Rolle 77 :-))

Rrg11, 2015-03-22:
Hello,

I don't mean to hijack this request but I'd second it, I would like to have support for this device too, although I don't wish to replace anything. If there's anything I could do to support developement for this model please let me know.

efidrin, 2015-04-09:
I have the same device (TL-WR841ND V9).
And I have replaced RAM (to 64mb) and flash (to 16 mb).
U-Boot flash original from TL-WR841ND V9 backup .
Then flash OpenWRT BB.
But have problem, cant write ART from backup.
Bootloader see only 4Mb flsh , and dont want write the last 64k from 16m.
You guys try some versions of MODED U-Boot on this device ?

ovctvct, 2015-12-19:
I also have this router, but I don't want to change any hardware. But please make it clear: does this mod work on v9 or not? If not, are you going to add support for it?

pepe2k, 2015-12-20:
Hello @ovctvct,

Current version of this project is not compatible with your device.
Next version will contain images for TL-WR841 v9.

Cheers,
Piotr

pepe2k, 2016-03-22:
New code has support for QCA953x.

Cheers,
Piotr

closed 2016-03-22 by pepe2k