pepe2k/u-boot_mod #55: Carambol2 build fails for u-boot/httpd

assarbad, 2015-04-17:
Hi there, I'm using the toolchain built with buildroot as explained on the 8devices Wiki in the article: [Upgrading Carambola 2 bootloader](http://8devices.com/wiki/carambola:2:gettingstarted:bootloader-upgrade).

The toolchain is underneath `$HOME/buildroot/output` and instead of exporting the variables, I decided to pass them on the command line, like so:

```
make PATH="$HOME/buildroot/output/host/usr/bin:$PATH" BUILD_TOPDIR=$(pwd) STAGING_DIR=$(pwd)/tmp MAKECMD="make ARCH=mips CROSS_COMPILE=mips-linux-" 8devices_carambola2
```
The sources from your repository are in `$HOME/u-boot_mod` and that's from where I invoke the above `make` command.

I also left out the `--silent`.

The problem now is in `u-boot/httpd`, pasting the relevant lines from the make output at the bottom right before and after the compilation error:

```
make[2]: Leaving directory `/home/oliver/u-boot_mod/u-boot/rtc'
make -C `dirname httpd/libhttpd.a`
make[2]: Entering directory `/home/oliver/u-boot_mod/u-boot/httpd'
mips-linux-gcc  -O  -D__KERNEL__ -DTEXT_BASE=0x9F000000 -I/home/oliver/u-boot_mod/u-boot/include -fno-builtin -ffreestanding -nostdinc -isystem /home/oliver/buildroot/output/host/usr/lib/gcc/mips-buildroot-linux-uclibc/4.7.4/include -pipe  -DCONFIG_MIPS -D__MIPS__ -G 0 -mabicalls -fpic -msoft-float -mips32 -march=mips32 -mtune=mips32 -EB -fno-schedule-insns -fno-schedule-insns2 -Wall -Wstrict-prototypes -DCOMPRESSED_UBOOT=0  -c -o uip.o uip.c
mips-linux-gcc  -O  -D__KERNEL__ -DTEXT_BASE=0x9F000000 -I/home/oliver/u-boot_mod/u-boot/include -fno-builtin -ffreestanding -nostdinc -isystem /home/oliver/buildroot/output/host/usr/lib/gcc/mips-buildroot-linux-uclibc/4.7.4/include -pipe  -DCONFIG_MIPS -D__MIPS__ -G 0 -mabicalls -fpic -msoft-float -mips32 -march=mips32 -mtune=mips32 -EB -fno-schedule-insns -fno-schedule-insns2 -Wall -Wstrict-prototypes -DCOMPRESSED_UBOOT=0  -c -o uip_arch.o uip_arch.c
mips-linux-gcc  -O  -D__KERNEL__ -DTEXT_BASE=0x9F000000 -I/home/oliver/u-boot_mod/u-boot/include -fno-builtin -ffreestanding -nostdinc -isystem /home/oliver/buildroot/output/host/usr/lib/gcc/mips-buildroot-linux-uclibc/4.7.4/include -pipe  -DCONFIG_MIPS -D__MIPS__ -G 0 -mabicalls -fpic -msoft-float -mips32 -march=mips32 -mtune=mips32 -EB -fno-schedule-insns -fno-schedule-insns2 -Wall -Wstrict-prototypes -DCOMPRESSED_UBOOT=0  -c -o uip_arp.o uip_arp.c
mips-linux-gcc  -O  -D__KERNEL__ -DTEXT_BASE=0x9F000000 -I/home/oliver/u-boot_mod/u-boot/include -fno-builtin -ffreestanding -nostdinc -isystem /home/oliver/buildroot/output/host/usr/lib/gcc/mips-buildroot-linux-uclibc/4.7.4/include -pipe  -DCONFIG_MIPS -D__MIPS__ -G 0 -mabicalls -fpic -msoft-float -mips32 -march=mips32 -mtune=mips32 -EB -fno-schedule-insns -fno-schedule-insns2 -Wall -Wstrict-prototypes -DCOMPRESSED_UBOOT=0  -c -o httpd.o httpd.c
mips-linux-gcc  -O  -D__KERNEL__ -DTEXT_BASE=0x9F000000 -I/home/oliver/u-boot_mod/u-boot/include -fno-builtin -ffreestanding -nostdinc -isystem /home/oliver/buildroot/output/host/usr/lib/gcc/mips-buildroot-linux-uclibc/4.7.4/include -pipe  -DCONFIG_MIPS -D__MIPS__ -G 0 -mabicalls -fpic -msoft-float -mips32 -march=mips32 -mtune=mips32 -EB -fno-schedule-insns -fno-schedule-insns2 -Wall -Wstrict-prototypes -DCOMPRESSED_UBOOT=0  -c -o fs.o fs.c
In file included from fs.c:54:0:
fsdata.c:333:1: error: expected expression before ‘,’ token
make[2]: *** [fs.o] Error 1
make[2]: Leaving directory `/home/oliver/u-boot_mod/u-boot/httpd'
make[1]: *** [httpd/libhttpd.a] Error 2
make[1]: Leaving directory `/home/oliver/u-boot_mod/u-boot'
make: *** [8devices_carambola2] Error 2
```
When we look at `u-boot/httpd/fsdata.c` line 333 we see (including the line numbers here):

```
325 static const char data_style_css[] = {
326 /* HTTP Header */
327 0x48, 0x54, 0x54, 0x50, 0x2f, 0x31, 0x2e, 0x30, 0x20, 0x32, 0x30, 0x30, 0x20, 0x4f, 0x4b, 0x0d, 0x0a,
328 0x53, 0x65, 0x72, 0x76, 0x65, 0x72, 0x3a, 0x20, 0x75, 0x49, 0x50, 0x2f, 0x30, 0x2e, 0x39, 0x0d, 0x0a,
329 0x43, 0x6f, 0x6e, 0x74, 0x65, 0x6e, 0x74, 0x2d, 0x74, 0x79, 0x70, 0x65, 0x3a, 0x20, 0x74, 0x65, 0x78,
330 0x74, 0x2f, 0x63, 0x73, 0x73, 0x3b, 0x20, 0x63, 0x68, 0x61, 0x72, 0x73, 0x65, 0x74, 0x3d, 0x55, 0x54,
331 0x46, 0x2d, 0x38, 0x0d, 0x0a, 0x0d, 0x0a,
332 /* Page/File content */
333 , 0 };
```

Which seems to be the problem. The comma at the end of line 331 and the comma at the beginning of line 333 mean there are two commas separating the final 0 element from the rest of the array.

This is revision 9a415a7f4040cb63c1e1233546360d386f3bfafb

Thanks for reading.

assarbad, 2015-04-17:
The fix is trivial, but I'm not sure this merely fixes the compilation but hides another issue or so ...

When I remove the leading comma on line 333 the compilation succeeds and all seems to be in order. However, I'm not sure this is safe to flash.

```
make[2]: Entering directory `/home/oliver/u-boot_mod/u-boot/httpd'
mips-linux-gcc  -O  -D__KERNEL__ -DTEXT_BASE=0x9F000000 -I/home/oliver/u-boot_mod/u-boot/include -fno-builtin -ffreestanding -nostdinc -isystem /home/oliver/buildroot/output/host/usr/lib/gcc/mips-buildroot-linux-uclibc/4.7.4/include -pipe  -DCONFIG_MIPS -D__MIPS__ -G 0 -mabicalls -fpic -msoft-float -mips32 -march=mips32 -mtune=mips32 -EB -fno-schedule-insns -fno-schedule-insns2 -Wall -Wstrict-prototypes -DCOMPRESSED_UBOOT=0  -c -o uip.o uip.c
mips-linux-gcc  -O  -D__KERNEL__ -DTEXT_BASE=0x9F000000 -I/home/oliver/u-boot_mod/u-boot/include -fno-builtin -ffreestanding -nostdinc -isystem /home/oliver/buildroot/output/host/usr/lib/gcc/mips-buildroot-linux-uclibc/4.7.4/include -pipe  -DCONFIG_MIPS -D__MIPS__ -G 0 -mabicalls -fpic -msoft-float -mips32 -march=mips32 -mtune=mips32 -EB -fno-schedule-insns -fno-schedule-insns2 -Wall -Wstrict-prototypes -DCOMPRESSED_UBOOT=0  -c -o uip_arch.o uip_arch.c
mips-linux-gcc  -O  -D__KERNEL__ -DTEXT_BASE=0x9F000000 -I/home/oliver/u-boot_mod/u-boot/include -fno-builtin -ffreestanding -nostdinc -isystem /home/oliver/buildroot/output/host/usr/lib/gcc/mips-buildroot-linux-uclibc/4.7.4/include -pipe  -DCONFIG_MIPS -D__MIPS__ -G 0 -mabicalls -fpic -msoft-float -mips32 -march=mips32 -mtune=mips32 -EB -fno-schedule-insns -fno-schedule-insns2 -Wall -Wstrict-prototypes -DCOMPRESSED_UBOOT=0  -c -o uip_arp.o uip_arp.c
mips-linux-gcc  -O  -D__KERNEL__ -DTEXT_BASE=0x9F000000 -I/home/oliver/u-boot_mod/u-boot/include -fno-builtin -ffreestanding -nostdinc -isystem /home/oliver/buildroot/output/host/usr/lib/gcc/mips-buildroot-linux-uclibc/4.7.4/include -pipe  -DCONFIG_MIPS -D__MIPS__ -G 0 -mabicalls -fpic -msoft-float -mips32 -march=mips32 -mtune=mips32 -EB -fno-schedule-insns -fno-schedule-insns2 -Wall -Wstrict-prototypes -DCOMPRESSED_UBOOT=0  -c -o httpd.o httpd.c
mips-linux-gcc  -O  -D__KERNEL__ -DTEXT_BASE=0x9F000000 -I/home/oliver/u-boot_mod/u-boot/include -fno-builtin -ffreestanding -nostdinc -isystem /home/oliver/buildroot/output/host/usr/lib/gcc/mips-buildroot-linux-uclibc/4.7.4/include -pipe  -DCONFIG_MIPS -D__MIPS__ -G 0 -mabicalls -fpic -msoft-float -mips32 -march=mips32 -mtune=mips32 -EB -fno-schedule-insns -fno-schedule-insns2 -Wall -Wstrict-prototypes -DCOMPRESSED_UBOOT=0  -c -o fs.o fs.c
mips-linux-ar crv libhttpd.a uip.o uip_arch.o uip_arp.o httpd.o fs.o
a - uip.o
a - uip_arch.o
a - uip_arp.o
a - httpd.o
a - fs.o
make[2]: Leaving directory `/home/oliver/u-boot_mod/u-boot/httpd'
UNDEF_SYM=`mips-linux-objdump -x lib_generic/libgeneric.a common/libcommon.a board/ar7240/ap121/libap121.a cpu/mips/libmips.a cpu/mips/ar7240/libar7240.a lib_mips/libmips.a drivers/libdrivers.a net/libnet.a rtc/librtc.a httpd/libhttpd.a  |sed  -n -e 's/.*\(__u_boot_cmd_.*\)/-u\1/p'|sort|uniq`;\
                mips-linux-ld -Bstatic -T /home/oliver/u-boot_mod/u-boot/board/ar7240/ap121/u-boot.lds -Ttext 0x9F000000  -G 0 -static -n -nostdlib $UNDEF_SYM cpu/mips/start.o \
                --start-group lib_generic/libgeneric.a common/libcommon.a board/ar7240/ap121/libap121.a cpu/mips/libmips.a cpu/mips/ar7240/libar7240.a lib_mips/libmips.a drivers/libdrivers.a net/libnet.a rtc/librtc.a httpd/libhttpd.a  --end-group -L /home/oliver/buildroot/output/host/usr/lib/gcc/mips-buildroot-linux-uclibc/4.7.4 -lgcc \
                -Map u-boot.map -o u-boot
mips-linux-objcopy --gap-fill=0xff -O srec u-boot u-boot.srec
mips-linux-objcopy --gap-fill=0xff -O binary u-boot u-boot.bin
> Preparing U-Boot image "u-boot.img"...
./tools/mkimage -A mips -T firmware -C none \
                -a 0x9F000000 -e 0 \
                -n  "U-Boot 1.1.4 for ap121 board" \
                -d u-boot.bin u-boot.img
    Image Name: U-Boot 1.1.4 for ap121 board
       Created: Fri Apr 17 17:27:46 2015
    Image Type: MIPS Linux Firmware (uncompressed)
     Data Size: 153816 Bytes = 150.21 kB = 0.15 MB
  Load Address: 0x9F000000
   Entry Point: 0x00000000
make[1]: Leaving directory `/home/oliver/u-boot_mod/u-boot'
> Preparing 256KB file filled with 0xFF...
> Copying U-Boot image...
> U-Boot image ready, size: 262144 bytes
```

assarbad, 2015-04-17:
I decided to take the risk and flash the resulting bootloader image and I can report success. The features I tested, including `httpd` seem to work.

The net console I started, but it locked up the serial console. I did not check any further.

pepe2k, 2015-04-18:
Hi @assarbad,

What toolchain exactly did you use? I don't see any information about toolchain on page you gave a link to.

Regards

assarbad, 2015-04-19:
@pepe2k apologies, I used the instructions from the Caraboot README https://github.com/8devices/Caraboot

pepe2k, 2015-04-27:
Hi @assarbad,

fsdata.c is generated automatically by this script: https://github.com/pepe2k/u-boot_mod/blob/master/u-boot/httpd/vendors/makefsdatac

Your fsdata.c looks bad - as you can see, there is only "HTTP Header" part and the rest (body of CSS file) is missing. Could you show me please whole fsdata.c file you got after compilation?

Do you have **Java RE** in your system? CSS files are compressed using **YUI Compressor**:
https://github.com/pepe2k/u-boot_mod/blob/master/u-boot/httpd/vendors/makefsdatac#L70

I think I should add there checking for Java available in system.

You can also do a small test (from top directory of source code) and check if there are some errors during the above script execute:

`cd u-boot/httpd`
`./vendors/makefsdatac general`
`cat fsdata.c`

Cheers,
Piotr

closed 2016-06-15 by pepe2k