pepe2k/u-boot_mod #57: git repo contains pre-compiled i386 ./host_util/lzma executable

frantony, 2015-04-20:
u-boot_mod relies on pre-compiled i386 ./host_util/lzma executable. Can we use host lzma or complile lzma on demand?

pepe2k, 2015-04-27:
Hello @frantony,

I will check it.

Cheers,
Piotr

frantony, 2015-04-28:
I have just checked lzma compression utilities from Debian for creating loadable image  ```6F01A8C0.img``` for TP-Link MR3020.

### lzma (usable)
lzma from lzma-9.22-2 package (```/usr/bin/lzma -> /etc/alternatives/lzma -> /usr/bin/lzmp```) with ```-c -k``` options is usable for ```6F01A8C0.img``` image creation.

### xz-utils (unusable)
lzma from xz-utils-5.1.1alpha+20120614-2+b3 package (```/usr/bin/lzma -> /etc/alternatives/lzma -> /usr/bin/xz```) with ```-c -k``` can't create usable image:
````
hornet> bootm 0x81000000
## Booting image at 81000000 ...
   Uncompressing Kernel Image ... Stream with EOS marker is not supportedLZMA ERROR
````

closed 2016-08-31 by pepe2k