pepe2k/u-boot_mod #58: Verification instructions request

hryamzik, 2015-04-22:
I have a 702n router and I've tried to replace it's original 2MB chip with an 8MB one with a new u-boot. It didn't work: power LED ignited at half brightness, network interface didn't go up. I soldered the original chip back it worked.

At this point I've no idea what went wrong. It could be the whole image or the u-boot part. @pepe2k, do you have any ideas about the debugging process?

Here's how I've prepared the 8MB image.

##### Compile u-boot

```
git clone https://github.com/pepe2k/u-boot_mod.git
cd u-boot_mod
ln -vs ~/aa/attitude_adjustment/staging_dir/toolchain-mips_r2_gcc-4.6-linaro_uClibc-0.9.33.2 toolchain

newline=$(($(grep -no "export MAKECMD=" Makefile|head -1|cut -d ':' -f 1)+1))
awk "NR==${newline}{print \"export PATH:=\$(BUILD_TOPDIR)/toolchain/bin/:\$(PATH)\"}7" Makefile|tee Makefile| head -7

sed -Ei 's/(mips-)(linux-)gnu-/\1openwrt-\2uclibc-/' Makefile

make tplink_wr702n
```

Here's the md5 if it could make any sense: ```3eeecc9c1a2654e00aee6876f0dc23ed *uboot_for_tp-link_tl-wr703n.bin```

##### Extract original image

```
mkdir -vp 702.orig

dd bs=1 skip=0                     count=$((64*1024))            if=702n.bin of=702.orig/uboot.bin
dd bs=1 skip=$((64*1024))          count=$((64*1024))            if=702n.bin of=702.orig/data.bin
dd bs=1 skip=$((128*1024))         count=$(((2*1024-64*3)*1024)) if=702n.bin of=702.orig/factory.bin
dd bs=1 skip=$(((2*1024-64)*1024))                               if=702n.bin of=702.orig/art.bin
```

##### Prepare the new, 8MB image

```
dd bs=1 count=$((64*1024))                            of=702n8m.bin if=uboot/uboot_for_tp-link_tl-wr703n.bin
dd bs=1 count=$((64*1024)) seek=$((64*1024))          of=702n8m.bin if=702.orig/data.bin
dd bs=1 count=$((64*1024)) seek=$((128*1024))         of=702n8m.bin if=702.orig/factory.bin
dd bs=1 count=$((64*1024)) seek=$(((8*1024-64)*1024)) of=702n8m.bin if=702.orig/art.bin
```

I've tried to mount extracted partitions, none worked. At this point I'm rally lost as all this maths is based on some internet knowledge and I've no idea how to verify partitions and u-boot image.

pepe2k, 2015-04-27:
Hello @hryamzik,

> make tplink_wr702n

Where did you get that from? My modification doesn't include make command for **WR702N** model.

Please, try my prebuild image for **WR703N** from last release: https://github.com/pepe2k/u-boot_mod/releases/tag/2014-11-19

Cheers,
Piotr

closed 2015-04-29 by pepe2k

hryamzik, 2015-05-05:
@pepe2k, that's a typo, sorry, I've done this for 703n. 702n is the same router as 703n one. It misses a few smd parts (related to USB) and a USB port and has a smaller ROM. Will give this image a try, but I don't think the issue is related to u-boot itself.

hryamzik, 2015-06-08:
Ok, got it working. Here's the correct way to reassemble the image:
```bash
v=703 sizeorig=4 sizenew=8
dd bs=1 skip=0                         count=$((128*1024))                  if=${v:-702}n.bin of=${v:-702}.orig/uboot.bin
dd bs=1 skip=$((128*1024))             count=$(((sizeorig*1024-64*3)*1024)) if=${v:-702}n.bin of=${v:-702}.orig/firmware.bin
dd bs=1 skip=$(((sizeorig*1024-64)*1024))                                   if=${v:-702}n.bin of=${v:-702}.orig/art.bin

mkdir -vp out
dd bs=1                                               of=out/${v:-702}n${sizenew}m.bin if="$uboot"
dd bs=1 conv=notrunc seek=$((128*1024))               of=out/${v:-702}n${sizenew}m.bin if=${v:-702}.orig/firmware.bin
dd bs=1 conv=notrunc seek=$(((sizenew*1024-64)*1024)) of=out/${v:-702}n${sizenew}m.bin if=${v:-702}.orig/art.bin
```