pepe2k/u-boot_mod #64: Support SKW71

ondonadas, 2015-05-11:
hi pepe2k

Recently I've buyed a  SKW71 module from SKYLAB (it cost 15$+ shipment=30$ as you say here https://github.com/pepe2k/u-boot_mod/issues/37). This is the manufacturer's "datasheet" http://www.skylab.com.cn/ZDXUpLoadFiles/2014060711252729.pdf

I would like to add uboot and kernel support for it. I've read openwrt guide http://wiki.openwrt.org/doc/devel/add.new.device in order  to add support for new devices. Also I have contacted manufacturer claiming for source code (the module comes preloaded with a openwrt rootfs).

Please could you recomend me how to go further in the uboot and kernel support?

Sergios
PS: thank you for share your work!

ondonadas, 2015-05-11:
Hi pepe2k

I've found this chinese link http://item.taobao.com/item.htm?spm=2013.1.20141002.4.q7OshF&scm=1007.10009.6098.i44610338154&id=44719631047&pvid=8158a966-05bd-4a9f-af81-243c4cc16fe9. After activate online translator it's  showing that SKW71 it's compatible with 703N SDK. I suposse that it makes reference to TP-Link TL-WR703N. 

Does it make sense for you?

Sergios

ondonadas, 2015-05-11:
SDK for SWK71 https://kanbox.com/f/RG0UO

hackruu, 2015-05-14:
who needs this hardware, while there's much better http://www.black-swift.com/?

pepe2k, 2015-06-01:
Hello @ondonadas,

If this module is compatible with WR703N, you can try image dedicated for WR703N. Then, if it works, you can take a look at any commit with support for other devices and make it the same way, for example: https://github.com/pepe2k/u-boot_mod/commit/d46bef51bd03f8a0a6429eab6f1acc9d04c73e40

Cheers

ondonadas, 2015-06-30:
Thanks @pepe2k,

finally I've change to Dragino module. 
Thank you for your interest (and for share your work).

Cheers

pepe2k, 2015-07-01:
Hello @ondonadas,

Good choice!

Cheers

closed 2015-07-01 by pepe2k