pepe2k/u-boot_mod #65: tplink_wdr3600_43x0 build error

frantony, 2015-05-25:
as commit a404a56e48c2796e4621b42cd461bf5e95ea5e09 (Enable loadb/loady commands by default for most of supported devices) ```make tplink_wdr3600_43x0``` leads to this build error:
````
common/libcommon.a(cmd_nvedit.o): In function `_do_setenv':
cmd_nvedit.c:(.text+0x6a4): undefined reference to `serial_setbrg'
cmd_nvedit.c:(.text+0x6a0): undefined reference to `serial_setbrg'
common/libcommon.a(cmd_load.o): In function `switch_baudrate':
cmd_load.c:(.text+0x280): undefined reference to `serial_setbrg'
cmd_load.c:(.text+0x284): undefined reference to `serial_setbrg'
Makefile:195: recipe for target 'u-boot' failed
make[1]: *** [u-boot] Error 1
Makefile:112: recipe for target 'tplink_wdr3600_43x0' failed
make: *** [tplink_wdr3600_43x0] Error 2
````

frantony, 2015-05-26:
Several other goals have the same problem:

* tplink_wa830re_v2_wa801nd_v2
* tplink_wr841n_v8
* tplink_mr3420_v2
* tplink_wdr3500

pepe2k, 2015-05-31:
Hello @frantony,

Support for AR934x is currently broken, it's missing new UART driver. Try code before commit https://github.com/pepe2k/u-boot_mod/commit/38f59a043532c43ef6e6d145edba2640f63375fb

Cheers

closed 2015-05-31 by pepe2k