pepe2k/u-boot_mod #66: Failed to compile for tplink_wr841n_v8

zz090923610, 2015-06-08:
Hi I'm compiling your uboot for tplink_wr841n_v8 from source, but failed with below error, could you please help me find out how to get over it? Thanks~
```
common/libcommon.a(cmd_nvedit.o): In function `_do_setenv':
cmd_nvedit.c:(.text+0x708): undefined reference to `serial_setbrg'
cmd_nvedit.c:(.text+0x704): undefined reference to `serial_setbrg'
common/libcommon.a(cmd_load.o): In function `switch_baudrate':
cmd_load.c:(.text+0x278): undefined reference to `serial_setbrg'
cmd_load.c:(.text+0x27c): undefined reference to `serial_setbrg'
make[1]: *** [u-boot] Error 1
make: *** [tplink_wr841n_v8] Error 2

pepe2k, 2015-06-08:
https://github.com/pepe2k/u-boot_mod/issues/65#issuecomment-107178105

closed 2015-06-08 by pepe2k

zhang3, 2015-06-11:
I uncounted this issues with ar9344/tp-link wdr4310 too. I think it is useless for changing band-rate in most case. so, for a dirty workaround, just remove it.
//serial_setbrg