pepe2k/u-boot_mod #68: WR841N V8.x ART partition problem

GBert, 2015-06-14:
First of all: many thx for your awesome work !

I facing a problem with the ART partition after upgrading to 16MB (W25Q128). The ART
was flashed backed on the last 64k, but UBoot claims an error:
```
*********************************************
*   U-Boot 1.1.4  (Apr 22 2015, 12:51:39)   *
*********************************************

AP123 (AR9341) U-Boot for TL-WR841N/D v8

DRAM:   32 MB DDR 32-bit
FLASH:  Winbond W25Q128 (16 MB)
CLOCKS: 535/400/200/25 MHz (CPU/RAM/AHB/SPI)

## Error: MAC is invalid, using fixed!

LED on during eth initialization...

## Error: MAC address in FLASH is invalid, using fixed!
## Error: MAC address in FLASH is invalid, using fixed!
Hit any key to stop autobooting:  0 

uboot> 
``` 
Here the flash layout:
```
root@OpenWrt:/# dmesg | grep " : "
[    0.740000] 0x000000000000-0x000000020000 : "u-boot"
[    0.750000] 0x000000020000-0x000000134e3c : "kernel"
[    0.750000] 0x000000134e3c-0x000000ff0000 : "rootfs"
[    0.770000] 0x000000470000-0x000000ff0000 : "rootfs_data"
[    0.780000] 0x000000ff0000-0x000001000000 : "art"
[    0.780000] 0x000000020000-0x000000ff0000 : "firmware"
```
and the content of the ART partition (OpenWRT trunk):
```
root@OpenWrt:/# hexdump -C /dev/mtd4
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00001000  02 02 00 02 03 04 05 06  00 30 3a 30 32 3a 30 33  |.........0:02:03|
00001010  3a 30 34 3a 30 35 3a 30  36 00 00 00 00 00 1f 00  |:04:05:06.......|
00001020  33 02 00 00 00 00 04 00  70 00 6d 00 03 00 08 ff  |3.......p.m.....|
00001030  61 01 00 00 00 10 01 00  00 22 22 02 00 50 01 50  |a........""..P.P|
00001040  01 50 01 00 00 00 00 00  00 1e 00 64 7d 96 a4 af  |.P.........d}...|
00001050  ff 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00001060  0e 0e 03 00 2c e2 00 02  0e 1c e0 e0 00 0c e0 e0  |....,...........|
00001070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00001080  00 00 00 00 00 00 00 00  00 00 70 89 a2 ff 00 8b  |..........p.....|
00001090  00 00 00 fe 00 8c 00 00  00 fc 00 8c 00 00 00 00  |................|
000010a0  00 95 00 00 00 ff 00 96  00 00 00 fe 00 95 00 00  |................|
000010b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000010c0  00 00 00 70 ac 70 89 ac  70 89 ac 70 89 ac 24 24  |...p.p..p..p..$$|
000010d0  24 24 24 24 24 24 24 22  20 1e 24 22 20 1e 24 22  |$$$$$$$" .$" .$"|
000010e0  20 1e 24 22 22 20 20 1c  20 1c 1c 1c 20 20 1c 14  | .$""  . ...  ..|
000010f0  24 22 22 20 20 1c 20 1c  1c 1c 20 20 1c 14 24 22  |$""  . ...  ..$"|
00001100  22 20 20 1c 20 1c 1c 1c  20 20 1c 14 24 22 22 20  |"  . ...  ..$"" |
00001110  20 1c 20 1c 1c 1c 20 20  1c 14 24 22 22 20 20 1c  | . ...  ..$""  .|
00001120  20 1c 1c 1c 20 20 1c 14  24 22 22 20 20 1c 20 1c  | ...  ..$""  . .|
00001130  1c 1c 20 20 1c 14 11 12  15 17 41 42 45 47 31 32  |..  ......ABEG12|
00001140  35 37 70 75 7a a2 70 75  7a a2 70 75 7a a2 7a 7f  |57puz.puz.puz.z.|
00001150  84 98 70 75 ac b8 70 75  ac b8 70 75 ac b8 70 75  |..pu..pu..pu..pu|
00001160  ac b8 70 75 7a ac 70 75  8e ac 70 75 8e ac 70 75  |..puz.pu..pu..pu|
00001170  ac a2 5a 5a 5a 1a 59 5f  5f 19 56 5e 5e 16 54 5e  |..ZZZ.Y__.V^^.T^|
00001180  5e 14 3c 7c 3c 7c 3c 7c  3c 7c 3c 7c 3c 7c 3c 7c  |^.<|<|<|<|<|<|<||
00001190  3c 7c 58 58 58 18 58 58  58 18 58 58 58 18 58 58  |<|XXX.XXX.XXX.XX|
000011a0  58 18 10 01 00 00 22 22  02 00 00 00 00 00 00 00  |X.....""........|
000011b0  00 00 00 00 00 00 44 00  00 00 00 00 00 ff 00 00  |......D.........|
000011c0  00 00 00 00 00 00 00 00  00 00 00 00 00 0e 0e 03  |................|
000011d0  00 2d e2 00 02 0e 1c 00  00 00 00 00 00 00 00 00  |.-..............|
000011e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000011f0  00 00 00 00 00 00 00 4c  54 68 78 8c a0 b9 cd 00  |.......LThx.....|
00001200  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00001280  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 4c  |...............L|
00001290  54 68 78 8c a0 b9 cd 4c  58 68 8c b4 bd b9 cd 4c  |Thx....LXh.....L|
000012a0  58 68 8c b4 bd b9 cd 14  14 14 0a 14 14 14 0a 14  |Xh..............|
000012b0  14 14 0a 14 14 14 0a 14  14 14 0a 14 14 14 0a 14  |................|
000012c0  14 14 0a 14 14 14 0a 14  14 0a 0a 00 00 0a 0a 00  |................|
000012d0  00 0a 0a 00 00 14 14 0a  0a 00 00 0a 0a 00 00 0a  |................|
000012e0  0a 00 00 14 14 0a 0a 00  00 0a 0a 00 00 0a 0a 00  |................|
000012f0  00 14 14 0a 0a 00 00 0a  0a 00 00 0a 0a 00 00 14  |................|
00001300  14 0a 0a 00 00 0a 0a 00  00 0a 0a 00 00 14 14 0a  |................|
00001310  0a 00 00 0a 0a 00 00 0a  0a 00 00 14 14 0a 0a 00  |................|
00001320  00 0a 0a 00 00 0a 0a 00  00 14 14 0a 0a 00 00 0a  |................|
00001330  0a 00 00 0a 0a 00 00 14  14 0a 0a 00 00 0a 0a 00  |................|
00001340  00 0a 0a 00 00 14 14 0a  0a 00 00 0a 0a 00 00 0a  |................|
00001350  0a 00 00 14 14 0a 0a 00  00 0a 0a 00 00 0a 0a 00  |................|
00001360  00 14 14 0a 0a 00 00 0a  0a 00 00 0a 0a 00 00 14  |................|
00001370  14 0a 0a 00 00 0a 0a 00  00 0a 0a 00 00 14 14 0a  |................|
00001380  0a 00 00 0a 0a 00 00 0a  0a 00 00 14 14 0a 0a 00  |................|
00001390  00 0a 0a 00 00 0a 0a 00  00 14 14 0a 0a 00 00 0a  |................|
000013a0  0a 00 00 0a 0a 00 00 10  16 18 40 46 48 30 36 38  |..........@FH068|
000013b0  4c 54 68 78 8c a0 b9 cd  4c 54 68 78 8c a0 b9 cd  |LThx....LThx....|
*
000013f0  4c 54 68 78 8c a0 b9 cd  3c 7c 3c 7c 3c 7c 3c 7c  |LThx....<|<|<|<||
00001400  3c 7c 3c 7c 3c 7c 3c 7c  3c 7c 3c 7c 3c 7c 3c 7c  |<|<|<|<|<|<|<|<||
*
00001440  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00010000
```

I'm not using the latest UBoot because of the UART problem in the newer versions.
Are these problems still exist ?

Do you have any hint regarding the ART recognition problem ?

Regards

Gerd

pepe2k, 2015-06-14:
Hello Gerd,

Your problem is not related with ART partition. MAC address in most of TP-Link devices is stored in second 64 KB sector of FLASH. I suppose that you wrote my U-Boot modification image on first sector, without restoring also second sector from original FLASH content.

Cheers

GBert, 2015-06-14:
Thx a lot ! Restored 2nd sector and it's working like a charme !

closed 2015-06-14 by GBert