pepe2k/u-boot_mod #72: Error: MDIO_BUSY! after ESD test

ondonadas, 2015-07-28:
Hi pepe2k

I'm using Dragino HE module and after some test, I've noticed that I've lost access to the ethernet interface. Booting with serial line console shows MDIO_BUSY and MDC check failed on uboot time.

...
## Error: MDIO_BUSY!
## Error: MDIO_BUSY!
ag7240_mii_setup: MDC check failed
## Error: MDIO_BUSY!
## Error: MDIO_BUSY!

...
Playing with uboot menu:
dr_boot> ping 192.168.255.2
## Error: MDIO_BUSY!
## Error: MDIO_BUSY!
## Error: MDIO_BUSY!
## Error: unknown eth speed!
Using eth0 device
## Error: ping failed, host 192.168.255.2 is not alive!

I'm wondering if you know how to workarround this issue. 

Sergio

pepe2k, 2015-07-28:
Hello @ondonadas,

What kind of tests exactly have you made? Is the Ethernet working in kernel?
For me it looks like hardware issue.

Piotr

ondonadas, 2015-07-28:
Hi @Piotr

I'm testing electrostatic discharge (ESD) without contact (8KV air discharge). I'm not aware in which moment Ethernet stops to works (maybe ESD have kill Ethernet module inside AR9331). I've test all ethernet  passive components in my circuit and it looks right.

In the kernel Ethernet doesn't work. Even more it doesn't initialize the hardware. After some debug I can see that MDIO Control Register 0x0098 is returning MDIO_BUSY bit = 1 (bit31 page 293 AR9931.pdf).

For the Dragino 2.0V_OUT which powers ethernet, doens't output the correct voltage.

Really I don't know how to debug this issue ¿?

pepe2k, 2015-07-28:
Hello @ondonadas,

> In the kernel Ethernet doesn't work. Even more it doesn't initialize the hardware. After some debug I can see that MDIO Control Register 0x0098 is returning MDIO_BUSY bit = 1 (bit31 page 293 AR9931.pdf).

I forgot that (at least in OpenWrt) kernel isn't responsible for Ethernet initialization, so if the U-Boot isn't able to do it... you won't be able to use it in kernel too.

> For the Dragino 2.0V_OUT which powers ethernet, doens't output the correct voltage.

That's a really bad sign.

> Really I don't know how to debug this issue ¿?

If it's a hardware issue, I don't see many debug possibilities in software side. I suppose that HE module is based (as many other, similar modules) on AP121 reference design - you should be able to find it with uncle Google and take a look on the Ethernet transformer power supply part.

> I'm testing electrostatic discharge (ESD) without contact (8KV air discharge). I'm not aware in which moment Ethernet stops to works (maybe ESD have kill Ethernet module inside AR9331). I've test all ethernet passive components in my circuit and it looks right.

I think that you have killed Ethernet switch in AR9331. I have seen many AR9331 based devices with broken Ethernet, with working other, external Ethernet related parts (including transformers).

Piotr

ondonadas, 2015-07-28:
Unfortunately I think that you are right.

But still not sure if it can be due a software error (on my side). I've
updated openwrt firmware to barrier breaker (without uboot).

Maybe after put barrier breaker version i've to upgrade uboot too? It
doesn't makes sense to me... But i'm very confused about how is possible
that a hardware issue 'triggers' MDIO-BUSY flag on MDIO Control Register.

Please Piotr, could you help me to understand this?

>
> Hello @ondonadas,
>>
>> In the kernel Ethernet doesn't work. Even more it doesn't initialize the
hardware. After some debug I can see that MDIO Control Register 0x0098 is
returning MDIO_BUSY bit = 1 (bit31 page 293 AR9931.pdf).
>
> I forgot that (at least in OpenWrt) kernel isn't responsible for Ethernet
initialization, so if the U-Boot isn't able to do it... you won't be able
to use it in kernel too.
>>
>> For the Dragino 2.0V_OUT which powers ethernet, doens't output the
correct voltage.
>
> That's a really bad sign.
>>
>> Really I don't know how to debug this issue ¿?
>
> If it's a hardware issue, I don't see many debug possibilities in
software side. I suppose that HE module is based (as many other, similar
modules) on AP121 reference design - you should be able to find it with
uncle Google and take a look on the Ethernet transformer power supply part.
>>
>> I'm testing electrostatic discharge (ESD) without contact (8KV air
discharge). I'm not aware in which moment Ethernet stops to works (maybe
ESD have kill Ethernet module inside AR9331). I've test all ethernet
passive components in my circuit and it looks right.
>
> I think that you have killed Ethernet switch in AR9331. I have seen many
AR9331 based devices with broken Ethernet, with working other, external
Ethernet related parts (including transformers).

closed 2015-11-09 by ondonadas