pepe2k/u-boot_mod #73: New device: WR740N v2.1

davidcie, 2015-08-11:
Has a supported W25Q32BVSIG 4MB flash, based on Atheros AR7240. Will be glad to help add support if given some general idea of where to start as I have little experience outside of desktop programming.

pepe2k, 2015-08-12:
Hello @davidcie,

This project doesn't and won't support EOL products, sorry.

Cheers

closed 2015-08-12 by pepe2k

davidcie, 2015-08-12:
Fair enough, thanks for your reply! Still, would be great if you could outline the general process of getting u-boot_mod ported so that someone with enough skill and free time can backport it to older devices (plenty of those still around).

pepe2k, 2015-08-12:
The easiest way is to get some GPL code from the vendor (TP-Link does that and releases its modified version of U-Boot) and combine it with this modification. You can take a look in commits history and find part with AR934x support.

davidcie, 2015-08-12:
Excellent, thanks for the useful info! Will have a stab when time permits — even if nothing pull-able comes out of it, looking forward to learing something new.