pepe2k/u-boot_mod #78: Build for tplink_wdr3600_43x0 fails with non-default value for CFG_PLL_FREQ

cosash, 2015-09-25:
If CFG_PLL_FREQ is set to a value other than the default CFG_PLL_560_480_240, the compilation fails with:

board/ar7240/db12x/libdb12x.a(lowlevel_init_934x.o):/mnt/sda3/u-boot_mod/u-boot/board/ar7240/db12x/../common/lowlevel_init_934x.S:81: undefined reference to `CPU_PLL_NFRAC_40'
board/ar7240/db12x/libdb12x.a(lowlevel_init_934x.o):/mnt/sda3/u-boot_mod/u-boot/board/ar7240/db12x/../common/lowlevel_init_934x.S:82: undefined reference to `DDR_PLL_NFRAC_40'
board/ar7240/db12x/libdb12x.a(lowlevel_init_934x.o):/mnt/sda3/u-boot_mod/u-boot/board/ar7240/db12x/../common/lowlevel_init_934x.S:89: undefined reference to `CPU_PLL_NFRAC_25'
board/ar7240/db12x/libdb12x.a(lowlevel_init_934x.o):/mnt/sda3/u-boot_mod/u-boot/board/ar7240/db12x/../common/lowlevel_init_934x.S:90: undefined reference to `DDR_PLL_NFRAC_25'
Makefile:227: recipe for target 'bootstrap' failed

cosash, 2015-09-25:
These are the defines which compile successfully, all other are failing:

CFG_PLL_400_400_200
CFG_PLL_566_500_250
CFG_PLL_566_400_200
CFG_PLL_560_480_240
CFG_PLL_533_400_200

pepe2k, 2015-09-26:
Hello @cosash,

Those defines (and O/C for AR934x platform in general) are not supported in code.

Regards,
Piotr

closed 2015-09-26 by pepe2k