pepe2k/u-boot_mod #84: TFTP boot from remote server + DNS

ondonadas, 2015-11-09:
Hi @pepe2k 

I guess that you're the right person to answer these doubts. If possible I'll like to hear your thoughts about it. 

I would like to modify uboot in order to download (using http or tftp) and write a new openwrt firmware image, from a remote "link", something like  https://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/openwrt-15.05-ar71xx-generic-a02-rb-w300n-squashfs-factory.bin. As far I've tested dns resolv doesn't work for uboot. Seems that it lacks dns client.

To achieve this, I wonder which could be the preferred method (or none of these):
1) tftp and download firmware image from a tftp server (DNS lacks?)
2) using uIP framework download image from a webserver/http (uIP supports DNS).
3) better to don't mess with uboot, and prepare a minimal kernel with tcp/ip stack + dns to download and write firmware in a 3rd partition.

Thank you for let me know your opinion.

pepe2k, 2015-11-09:
Hello @ondonadas,

From my POV, bootloader should be responsible only for basic hardware initialization, configuration, maybe some verification (like memory test) and then start kernel. Plus, eventually gives possibility to re/write firmware and that's all.

For what you need, I would go with 3rd option. Of course, it should be possible to do it in U-Boot, probably using uIP (for dns: I have seen some patches with dns command for newer U-Boot versions), but remember that downloading binary images from HTTP or TFTP from **unknown**, remote servers and writing them directly to FLASH... that really doesn't sound **secure** for me.

I have seen some boards with dual-image support - one, full and second, smaller which U-Boot runs only when first one is broken. That second one (call it recovery) is responsible only for flashing first with a working image.

Cheers,
Piotr

ondonadas, 2015-11-09:
Thank you @pepe2k for your effor

I'm agree with you
Currently I've 2MB of free space. Hope that it will be enough!

cheers!
Sergio

closed 2015-11-13 by ondonadas