pepe2k/u-boot_mod #85: updating Dragino HE

ondonadas, 2015-11-13:
Hi @pepe2k 

Please If you're so kindly, I would like you to help me to diagnose where is the error(s) I did, updating Dragino HE uboot.

Basically I clone Barrier Breaker from openwrt git. After that I've configured openwrt/menuconfig to generate uboot "uboot-ar71xx-nbg460n_550n_550nh" (I've suppose that uboot on openwrt/git it's the same that it's available from github.com/pepe2k). 

Put that file to TFTP directory and powers Dragino.
BTW, actually Dragino HE MTD partitions looks like:
<pre>
cat /proc/mtd 
dev:    size   erasesize  name
mtd0: 00040000 00010000 "u-boot"
mtd1: 00fa0000 00010000 "firmware"
mtd2: 00120000 00010000 "kernel"
mtd3: 00e80000 00010000 "rootfs"
mtd4: 00c50000 00010000 "rootfs_data"
mtd5: 00010000 00010000 "config"
mtd6: 00010000 00010000 "art"
</pre>

<pre>
bootargs=console=ttyS0,115200 root=/dev/ram rootfstype=initramfs init=/sbin/init mtdparts=ar7240-nor0:192k(u-boot),64k(u-boot-env),16064k(firmware),64k(ART)
</pre>
Please note uboot size 192K = 0x30000

After that I've run this 3 commands:
> tftpboot 0x80800000 openwrt-ar71xx-generic-nbg460n_550n_550nh-u-boot.bin
> erase 0x9F000000 +0x30000
> cp.b 0x80800000 0x9F000000 0x30000

And "reset" => RIP :-O

This are full traces
<pre>
*********************************************
*        U-Boot 1.1.4  (Jun  3 2014)        *
*********************************************

AP121 (AR9331) U-Boot for Dragino v2 MS14

DRAM:   64 MB
FLASH:  Winbond W25Q128 (16 MB)
CLOCKS: 400/400/200/33 MHz (CPU/RAM/AHB/SPI)

LED on during eth initialization...

Hit any key to stop autobooting:  0

dr_boot> printenv
bootcmd=bootm 0x9F040000
bootdelay=1
baudrate=115200
bootfile="firmware.bin"
loadaddr=0x80800000
ncport=6666
uboot_addr=0x9F000000
uboot_name=uboot.bin
uboot_size=0x30000
uboot_upg=if ping $serverip; then tftp $loadaddr $uboot_name && if itest.l $filesize == $uboot_size; then erase $uboot_addr +$filesize && cp.b $loadaddr $uboot_addr $filesize && echo OK!; else echo ERROR! Wrong file size!; fi; else ERROR! Server not reachable!; fi
firmware_addr=0x9F040000
firmware_name=firmware.bin
firmware_upg=if ping $serverip; then tftp $loadaddr $firmware_name && erase $firmware_addr +$filesize && cp.b $loadaddr $firmware_addr $filesize && echo OK!; else ERROR! Server not reachable!; fi
ethact=eth0
serverip=192.168.255.100
ipaddr=192.168.255.1
netmask=255.255.255.0
bootargs=console=ttyS0,115200 root=/dev/ram rootfstype=initramfs init=/sbin/init mtdparts=ar7240-nor0:192k(u-boot),64k(u-boot-env),16064k(firmware),64k(ART)
stdin=serial
stdout=serial
stderr=serial

Environment size: 962/32764 bytes

dr_boot> tftpboot 0x80800000 openwrt-ar71xx-generic-nbg460n_550n_550nh-u-boot.bin
Ethernet mode (duplex/speed): 1/100 Mbps

TFTP from IP: 192.168.255.100
      Our IP: 192.168.255.1
    Filename: 'openwrt-ar71xx-generic-nbg460n_550n_550nh-u-boot.bin'
Load address: 0x80800000
       Using: eth0

     Loading: ################################

TFTP transfer complete!

Bytes transferred: 160200 (0x271c8)
dr_boot> cp.b
Usage:
cp [.b, .w, .l] source target count
        - copy memory

dr_boot> erase 0x9F000000 +0x30000
Erase FLASH from 0x9F000000 to 0x9F02FFFF in bank #1
Erasing: ###

Erased sectors: 3

dr_boot> cp.b 0x80800000 0x9F000000 0x30000
Copying to FLASH...
Writting at address: 0x9F000000

Done!

dr_boot> md 0x9F000000
9F000000: 100000FF 00000000 100000FD 00000000    ................
9F000010: 10000157 00000000 10000155 00000000    ...W.......U....
9F000020: 10000153 00000000 10000151 00000000    ...S.......Q....
9F000030: 1000014F 00000000 1000014D 00000000    ...O.......M....
9F000040: 1000014B 00000000 10000149 00000000    ...K.......I....
9F000050: 10000147 00000000 10000145 00000000    ...G.......E....
9F000060: 10000143 00000000 10000141 00000000    ...C.......A....
9F000070: 1000013F 00000000 1000013D 00000000    ...?.......=....
9F000080: 1000013B 00000000 10000139 00000000    ...;.......9....
9F000090: 10000137 00000000 10000135 00000000    ...7.......5....
9F0000A0: 10000133 00000000 10000131 00000000    ...3.......1....
9F0000B0: 1000012F 00000000 1000012D 00000000    .../.......-....
9F0000C0: 1000012B 00000000 10000129 00000000    ...+.......)....
9F0000D0: 10000127 00000000 10000125 00000000    ...'.......%....
9F0000E0: 10000123 00000000 10000121 00000000    ...#.......!....
9F0000F0: 1000011F 00000000 1000011D 00000000    ................
dr_boot> reset

Resetting the board...
</pre>

I'll like too know your thoughts.

Sergio

pepe2k, 2015-11-13:
Hello @ondonadas,

>[...]
> Basically I clone Barrier Breaker from openwrt git. After that I've configured openwrt/menuconfig to generate uboot "uboot-ar71xx-nbg460n_550n_550nh" (I've suppose that uboot on openwrt/git it's the same that it's available from github.com/pepe2k).
> [...]

Why did you use image dedicated for nbg460n/550n/550nh boards in your Dragino 2 board? It's not universal and, as you have already confirmed, won't work on other boards with different hardware.

My code is not included/available in OpenWrt repository and to be honest I don't understand why you decided to change factory U-Boot on Dragino 2? By default Dragino 2 uses my modification.

You board is bricked. To make it working again, in short words: desolder FLASH chip, use some FLASH programmer and put there proper U-Boot image, solder it back.

Cheers,
Piotr

ondonadas, 2015-11-13:
yeap! 

Simply I didn't understand that uboot inside openwrt is not universal or almost don't take care of menuconfig platform selection.

Big thanks to you @pepe2k

closed 2015-11-13 by ondonadas