pepe2k/u-boot_mod #90: Support for Arduino Yun and Arduino Tian

mimmoLinino, 2016-02-02:
Is there any possibilities to extend the support of this u-boot to Arduino boards?
[Arduino Yun](http://www.arduino.org/products/boards/4-arduino-boards/arduino-yun) is based on AR9331, [Arduino Tian](http://www.arduino.org/products/boards/4-arduino-boards/arduino-tian) is based on AR9342

plntyk, 2016-02-10:
mainline uboot might get support for ath79/ar933x soon once this series::
http://lists.denx.de/pipermail/u-boot/2016-January/241592.html

is accepted

pepe2k, 2016-03-22:
Hello,

@mimmoLinino, I don't have access to those boards, but it shouldn't be a big problem to make support for them as the SoCs are already supported.

@plntyk, sure, some day... I'm still waiting for proper copyrights handling in those patches, because lot of code there is just copy&past from my repository and CodeAurora...

Cheers,
Piotr

closed 2016-06-15 by pepe2k