pepe2k/u-boot_mod #95: {standard input}:1618: Error: operand 2 must be an immediate expression sw $4,$5($2)' {standard input}:1665: Error: operand 2 must be an immediate expressionsw $4,$5($2)'

stopspazzing, 2016-03-25:
Using 64bit debian
> spazz@debian:~/u-boot_mod$ make tplink_wdr3600_43x0
Makefile:154: warning: overriding recipe for target '.PHONY'
Makefile:152: warning: ignoring old recipe for target '.PHONY'
Makefile:240: warning: overriding recipe for target 'board/ar7240/ap121/libap121.a'
Makefile:208: warning: ignoring old recipe for target 'board/ar7240/ap121/libap121.a'
Makefile:240: warning: overriding recipe for target 'cpu/mips/libmips.a'
Makefile:208: warning: ignoring old recipe for target 'cpu/mips/libmips.a'
Makefile:240: warning: overriding recipe for target 'cpu/mips/ar7240/libar7240.a'
Makefile:208: warning: ignoring old recipe for target 'cpu/mips/ar7240/libar7240.a'
> Configuring for TP-Link TL-WDR3600/43x0 at: Thu Mar 24 19:20:14 PDT 2016 
Makefile:154: warning: overriding recipe for target '.PHONY'
Makefile:152: warning: ignoring old recipe for target '.PHONY'
Makefile:240: warning: overriding recipe for target 'board/ar7240/db12x/libdb12x.a'
Makefile:208: warning: ignoring old recipe for target 'board/ar7240/db12x/libdb12x.a'
Makefile:240: warning: overriding recipe for target 'cpu/mips/libmips.a'
Makefile:208: warning: ignoring old recipe for target 'cpu/mips/libmips.a'
Makefile:240: warning: overriding recipe for target 'cpu/mips/ar7240/libar7240.a'
Makefile:208: warning: ignoring old recipe for target 'cpu/mips/ar7240/libar7240.a'
r - crc32.o
r - ctype.o
r - display_options.o
r - ldiv.o
r - string.o
r - vsprintf.o
r - LzmaWrapper.o
r - LzmaDecode.o
In file included from /home/spazz/u-boot_mod/u-boot/include/cmd_qcaclk.h:13:0,
                 from cmd_qcaclk.c:13:
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1637:12: warning: inline function 'qca_dram_tras_lat' declared but never defined
 inline u32 qca_dram_tras_lat(void);
            ^
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1636:12: warning: inline function 'qca_dram_trp_lat' declared but never defined
 inline u32 qca_dram_trp_lat(void);
            ^
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1635:12: warning: inline function 'qca_dram_trcd_lat' declared but never defined
 inline u32 qca_dram_trcd_lat(void);
            ^
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1634:12: warning: inline function 'qca_dram_cas_lat' declared but never defined
 inline u32 qca_dram_cas_lat(void);
            ^
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1621:12: warning: inline function 'qca_xtal_is_40mhz' declared but never defined
 inline u32 qca_xtal_is_40mhz(void);
            ^
r - main.o
r - cmd_bootm.o
r - cmd_boot.o
r - cmd_flash.o
r - cmd_mem.o
r - cmd_custom.o
r - cmd_net.o
r - cmd_nvedit.o
r - cmd_itest.o
r - cmd_load.o
r - cmd_qcaclk.o
r - command.o
r - console.o
r - devices.o
r - dlmalloc.o
r - environment.o
r - env_common.o
r - env_nowhere.o
r - exports.o
r - flash.o
r - lists.o
r - env_flash.o
r - hush.o
r - xyzModem.o
r - crc16.o
r - s_record.o
In file included from db12x.c:10:0:
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1637:12: warning: inline function 'qca_dram_tras_lat' declared but never defined
 inline u32 qca_dram_tras_lat(void);
            ^
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1636:12: warning: inline function 'qca_dram_trp_lat' declared but never defined
 inline u32 qca_dram_trp_lat(void);
            ^
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1635:12: warning: inline function 'qca_dram_trcd_lat' declared but never defined
 inline u32 qca_dram_trcd_lat(void);
            ^
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1634:12: warning: inline function 'qca_dram_cas_lat' declared but never defined
 inline u32 qca_dram_cas_lat(void);
            ^
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1621:12: warning: inline function 'qca_xtal_is_40mhz' declared but never defined
 inline u32 qca_xtal_is_40mhz(void);
            ^
a - db12x.o
a - ../common/ar7240_pci.o
a - ../common/spi_flash.o
a - ../common/common.o
a - ../common/athrs17_phy.o
r - cpu.o
r - cache.o
In file included from qca_common.c:12:0:
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1637:12: warning: inline function 'qca_dram_tras_lat' declared but never defined
 inline u32 qca_dram_tras_lat(void);
            ^
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1636:12: warning: inline function 'qca_dram_trp_lat' declared but never defined
 inline u32 qca_dram_trp_lat(void);
            ^
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1635:12: warning: inline function 'qca_dram_trcd_lat' declared but never defined
 inline u32 qca_dram_trcd_lat(void);
            ^
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1634:12: warning: inline function 'qca_dram_cas_lat' declared but never defined
 inline u32 qca_dram_cas_lat(void);
            ^
In file included from qca_clocks.c:16:0:
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1637:12: warning: inline function 'qca_dram_tras_lat' declared but never defined
 inline u32 qca_dram_tras_lat(void);
            ^
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1636:12: warning: inline function 'qca_dram_trp_lat' declared but never defined
 inline u32 qca_dram_trp_lat(void);
            ^
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1635:12: warning: inline function 'qca_dram_trcd_lat' declared but never defined
 inline u32 qca_dram_trcd_lat(void);
            ^
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1634:12: warning: inline function 'qca_dram_cas_lat' declared but never defined
 inline u32 qca_dram_cas_lat(void);
            ^
/home/spazz/u-boot_mod/u-boot/include/soc/qca_soc_common.h:1621:12: warning: inline function 'qca_xtal_is_40mhz' declared but never defined
 inline u32 qca_xtal_is_40mhz(void);
            ^
{standard input}: Assembler messages:
{standard input}:1106: Error: operand 2 must be an immediate expression `lb $2,$4($2)'
{standard input}:1165: Error: operand 2 must be an immediate expression `lb $2,$4($2)'
{standard input}:1224: Error: operand 2 must be an immediate expression `lh $2,$4($2)'
{standard input}:1283: Error: operand 2 must be an immediate expression `lh $2,$4($2)'
{standard input}:1336: Error: operand 2 must be an immediate expression `lw $2,$4($2)'
{standard input}:1383: Error: operand 2 must be an immediate expression `lw $2,$4($2)'
{standard input}:1430: Error: operand 2 must be an immediate expression `sb $4,$5($2)'
{standard input}:1477: Error: operand 2 must be an immediate expression `sb $4,$5($2)'
{standard input}:1524: Error: operand 2 must be an immediate expression `sh $4,$5($2)'
{standard input}:1571: Error: operand 2 must be an immediate expression `sh $4,$5($2)'
{standard input}:1618: Error: operand 2 must be an immediate expression `sw $4,$5($2)'
{standard input}:1665: Error: operand 2 must be an immediate expression `sw $4,$5($2)'
{standard input}:1735: Error: operand 2 must be an immediate expression `lb $1,$4($2)'
{standard input}:1812: Error: operand 2 must be an immediate expression `lh $1,$4($2)'
{standard input}:1889: Error: operand 2 must be an immediate expression `lw $1,$4($2)'
{standard input}:1968: Error: operand 2 must be an immediate expression `sb $1,$4($2)'
{standard input}:2045: Error: operand 2 must be an immediate expression `sh $1,$4($2)'
{standard input}:2122: Error: operand 2 must be an immediate expression `sw $1,$4($2)'
/home/spazz/u-boot_mod/u-boot/config.mk:248: recipe for target 'qca_clocks.o' failed
make[2]: *** [qca_clocks.o] Error 1
Makefile:240: recipe for target 'cpu/mips/ar7240/libar7240.a' failed
make[1]: *** [cpu/mips/ar7240/libar7240.a] Error 2
Makefile:117: recipe for target 'tplink_wdr3600_43x0' failed
make: *** [tplink_wdr3600_43x0] Error 2

olewales, 2016-03-25:
Same here, using [toolchain](https://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/OpenWrt-Toolchain-ar71xx-generic_gcc-5.3.0_musl-1.1.14.Linux-x86_64.tar.bz2) linked up in README

This assembly code seems to be generated by macros in [u-boot/include/asm-mips/io.h](https://github.com/pepe2k/u-boot_mod/blob/master/u-boot/include/asm-mips/io.h#L223) but understanding those is beyond me.

pepe2k, 2016-03-29:
Fixed in https://github.com/pepe2k/u-boot_mod/commit/6d509b9219dc0e87f06ac2f24df60505fc1af707

closed 2016-03-29 by pepe2k