pepe2k/u-boot_mod #96: TL-WR841n v9 will not autoboot

ovctvct, 2016-03-25:
I just compiled and installed this mod with my tl-wr841n, but it will just not boot automatically, but running boot from netconsole it boots fine. Here's a netconsole log for referrence:
```
uboot> printenv
bootargs=console=ttyS0,115200 root=31:02 rootfstype=squashfs init=/sbin/init mtdparts=ath-nor0:128k(u-boot),1024k(kernel),2816k(rootfs),64k(config),64k(art)
bootcmd=bootm 0x9F020000
bootdelay=1
baudrate=115200
ipaddr=192.168.1.1
serverip=192.168.1.2
bootfile="firmware.bin"
loadaddr=0x80800000
ncport=6666
uboot_addr=0x9F000000
uboot_name=uboot.bin
uboot_size=0x1EC00
uboot_backup_size=0x20000
uboot_upg=if ping $serverip; then mw.b $loadaddr 0xFF $uboot_backup_size && cp.b $uboot_addr $loadaddr $uboot_backup_size && tftp $loadaddr $uboot_name && if itest.l $filesize <= $uboot_size; then erase $uboot_addr +$uboot_backup_size && cp.b $loadaddr $uboot_addr $uboot_backup_size && echo OK!; else echo ERROR! Wrong file size!; fi; else echo ERROR! Server not reachable!; fi
stdin=nc
stdout=nc
stderr=nc
ethact=eth1

Environment size: 814/4092 bytes

uboot> boot
Booting image at: 0x9F020000

   Image name:   OpenWrt r46767
   Image type:   MIPS Linux Kernel Image (lzma compressed)
   Data size:    1098588 Bytes = 1 MB
   Load address: 0x80060000
   Entry point:  0x80060000
```

pepe2k, 2016-03-25:
Can you provide me whole bootlog from serial console?

ovctvct, 2016-03-25:
Everything looks normal, before this:
`Booting image at: 0x9F020000

   Image name:   OpenWrt r46767
   Image type:   MIPS Linux Kernel Image (lzma compressed)
   Data size:    1098588 Bytes = 1 MB
   Load address: 0x80060000
   Entry point:  0x80060000
Uncompressing kernel image... ## Error: LZMA error num: 1`

Then web failsafe opens.

pepe2k, 2016-03-25:
I need whole bootlog output, from begin.

ovctvct, 2016-03-25:
```
***************************************
*     U-Boot 1.1.4-77ad4d95-dirty     *
*        Mar 23 2016, 16:30:33        *
***************************************

BOARD: TP-Link TL-WR841N v9
    SOC: QCA953x ver. 1 rev. 1
    CPU: MIPS 24Kc
    RAM: 32 MB DDR1 16-bit
  FLASH: 4 MB Spansion FL032PIF 
    MAC: -removed-
 CLOCKS: CPU/RAM/AHB/SPI/REF
         550/400/200/ 25/ 25 MHz

Hit any key to stop autoboot: 0  1

Booting image at: 0x9F020000

   Image name:   OpenWrt r46767
   Image type:   MIPS Linux Kernel Image (lzma compressed)
   Data size:    1098588 Bytes = 1 MB
   Load address: 0x80060000
   Entry point:  0x80060000

Uncompressing kernel image... ## Error: LZMA error num: 1

```

That's it.

pepe2k, 2016-03-25:
1. What toolchain did you use to build the code?
2. What changes did you make in the code?

ovctvct, 2016-03-25:
1.This toolchain(if you don't know, I am user5077 on openwrt forum): http://downloads.openwrt.org/attitude_adjustment/12.09/ar71xx/generic/OpenWrt-Toolchain-ar71xx-for-mips_r2-gcc-4.6-linaro_uClibc-0.9.33.2.tar.bz2

2. No changes, just compiled, and changed bootdelay via setenv, not source code.

anon95, 2016-03-26:
Same issue with a wdr4300. Immediately goes to console. Interestingly autoboot works if UART TX is connected.

Sudokamikaze, 2016-03-26:
Same issue with TL-WR841ND V8. if i boot to network console and write "boot" bootloader will boot openwrt firmware

pepe2k, 2016-03-31:
Problem should be fixed in https://github.com/pepe2k/u-boot_mod/commit/247b1d69137d991af406d854d14ae039585ae096.

For now it's more like a workaround, because the real source of the problem still needs investigation. It seems that the SoC just after configuring LS UART and printing first character on TX line reads NULL character on RX line when it's not pulled up. It looks like a break interrupt, but the BI bit in line status register (LSR) is not set.

closed 2016-03-31 by pepe2k

ovctvct, 2016-03-31:
Well, the issue still happens to me, but I don't have the serial TX problem, it works fine, just doesn't want to autoboot. Interestingly openwrt failsafe seems to work, but it's not accessible from network.

pepe2k, 2016-03-31:
@ovctvct your problem is not related with the LS UART. Give me please link to the exact OpenWrt image which is not working for you.

ovctvct, 2016-03-31:
OK,  https://drive.google.com/file/d/0B2CQR48oiPa5WFAtLUxGUGVRNGc/view?usp=sharing

pepe2k, 2016-03-31:
@ovctvct

> It's a custom build DD snapshot with u-boot unlocked. The image boots fine with the factory u-boot. With this u-boot only if run boot manually it boots.

You have some problems with your image and/or build system. I don't know what kind of changes did you make in image, so all I can suggest you is to rebuild everything, including the U-Boot (pull all my last changes and use toolchain from your OpenWrt build).

Your image from the link boots (automatically) on my TL-WR841Nv9 without any problems, but the kernel hangs:

```
***************************************
*     U-Boot 1.1.4-247b1d69-clean     *
*        Mar 31 2016, 12:02:56        *
***************************************

** Warning: bad env CRC, using default,
   use 'saveenv' to save it in FLASH

  BOARD: TP-Link TL-WR841N/D v9
    SOC: QCA953x ver. 1 rev. 1
    CPU: MIPS 24Kc
    RAM: 32 MB DDR1 16-bit CL3-3-3-8
  FLASH: 4 MB Macronix MX25L320
    MAC: XX:XX:XX:XX:XX:XX
 CLOCKS: CPU/RAM/AHB/SPI/REF
         550/400/200/ 25/ 25 MHz

Hit any key to stop autoboot: 0

Booting image at: 0x9F020000

   Image name:   OpenWrt r49081
   Image type:   MIPS Linux Kernel Image (lzma compressed)
   Data size:    1163144 Bytes = 1.1 MB
   Load address: 0x80060000
   Entry point:  0x80060000

Uncompressing kernel image... OK!
Starting kernel...

[    0.000000] Linux version 4.1.20 (ovctvct@ovctvct-virtual-machine) (gcc version 5.3.0 (OpenWrt GCC 5.3.0 r49081) ) #1 Wed Mar 23 17:22:03 UTC 2016
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] SoC: Qualcomm Atheros QCA9533 ver 1 rev 1
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 02000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000001ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000001ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000001ffffff]
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
[    0.000000] Kernel command line:  board=TL-WR841N-v9  console=ttyS0,115200 rootfstype=squashfs,jffs2 noinitrd
[    0.000000] PID hash table entries: 128 (order: -3, 512 bytes)
[    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 28264K/32768K available (2560K kernel code, 133K rwdata, 652K rodata, 228K init, 188K bss, 4504K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:83
[    0.000000] Clocks: CPU:550.000MHz, DDR:400.000MHz, AHB:200.000MHz, Ref:25.000MHz
[    0.000000] clocksource MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6950037990 ns
[    0.000010] sched_clock: 32 bits at 275MHz, resolution 3ns, wraps every 7809031678ns
[    0.008019] Calibrating delay loop... 366.18 BogoMIPS (lpj=1830912)
[    0.070702] pid_max: default: 32768 minimum: 301
[    0.075588] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[... 152 lines cut ...]
[    8.821076] procd: - init -
Please press Enter to activate this console.
[    9.715386] ip6_tables: (C) 2000-2006 Netfilter Core Team
[    9.737977] Loading modules backported from Linux version v4.4-rc5-1913-gc8fdf68
[    9.745697] Backport generated by backports.git backports-20151218-0-g2f58d9d
[    9.756890] ip_tables: (C) 2000-2006 Netfilter Core Team
[    9.774896] nf_conntrack version 0.5.0 (438 buckets, 1752 max)
[    9.822018] xt_time: kernel timezone is -0000
[    9.893517] PPP generic driver version 2.4.2
[    9.901049] NET: Registered protocol family 24
[    9.973424] ieee80211 phy0: Atheros AR9531 Rev:1 mem=0xb8100000, irq=47
[   17.582798] jffs2_scan_eraseblock(): End of filesystem marker found at 0x0
[   17.618532] jffs2_build_filesystem(): unlocking the mtd device... done.
[   17.625389] jffs2_build_filesystem(): erasing all blocks after the end marker... done.
[   22.715145] jffs2: notice: (943) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[   22.750404] device eth0 entered promiscuous mode
[   22.769665] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   22.868180] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   24.351062] eth0: link up (1000Mbps/Full duplex)
[   24.355909] br-lan: port 1(eth0) entered forwarding state
[   24.361566] br-lan: port 1(eth0) entered forwarding state
[   24.433118] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   26.358512] br-lan: port 1(eth0) entered forwarding state
[...]
```

ovctvct, 2016-03-31:
With my latest uboot image I did not try other firmware than my own compiled one. So it could easily be a firmware issue. Here's the u-boot I compiled(and I used openwrt toolchain) anyway:  https://drive.google.com/file/d/0B2CQR48oiPa5bGFyX19JUzlHQlk/view?usp=sharing

mikebdp2, 2019-10-01:
Unfortunately this autoboot problem is still happening with the latest u-boot_mod built for a TL-WR841ND v9. I am not sure yet at exactly what stage it gets stuck _(need to learn more about the logs extraction on this model, please advise if you could)_, however it's booting fine with TP-Link's u-boot. 

I have soldered a SOIC-8 socket on this router for a super easy removal/extraction of a chip _(ISP flashing with a test clip doesn't work on this model),_ so - if you have any experimental builds - we could debug this issue together.

mikebdp2, 2019-10-01:
@pepe2k please reopen this bug

mikebdp2, 2019-10-01:
@pepe2k @ovctvct please reopen this bug

phizev, 2019-10-02:
@mikebdp2 while possibly not directly related, I have run into this issue only when connected to the serial console on the board. Sometimes as the router was booting, it would pick up stray input from the serial console and then halt there. Mainly an issue when it was hard powered on while connected to an RS232 adapter, then several meters of cable to a PC.

Not an issue if the PC was disconnected, or it was rebooted from software.

mikebdp2, 2020-02-05:
@phizev I got disappointed at u-boot_mod because of this problem, and - since that a bootloader containing a web server may be not a security-wise idea - perhaps I'm not going to investigate. Instead, Currently I'm trying to upgrade a TP-Link's ancient U-Boot to a newer U-Boot versions until I reach a version where the official DENX "Das U-Boot" repository got a support for 16MB flash chips. 

I've already learned at what points of history TP-Link has forked U-Boot _(some parts of repo were newer than others, so a bit complicated!)_, then discovered the custom TP-Link patches which added a support for TL-WR841N(D) v9, and now rebasing these patches on top of more and more newer DENX U-Boot versions. It's a slow and boring process because of many merge conflicts and compilation errors, but I've already reached a version v1.3.3 - https://github.com/mikebdp2/wr841nd_v9_u-boot_r3b0rn_v133a

Currently I'm having a small break, so if anyone would like to join me - you're welcome! First of all you'll have to solder a SOIC8 socket like https://www.thelabeshop.com/en/Programmer-Emulators/SPI-Flash-Programmers/Accessories/Dediprog-SPI-Flash-Socket-8-Pin/ to your router instead of the old flash chip (which you could remove either by soldering or using a dremel-like tool to carefully saw off its' legs and then desolder them individually).

phizev, 2021-11-21:
As an update to my specific situation, and resolution, I switched to using a shielded cable from the serial pin out to my RS232 adapter/breakout board. Cable was half a shielded CD-ROM audio cable, RS232 adapter/breakout board has been permanently installed for close to 2 years now with no further issues booting.

In my case, I feel comfortable that it was just stray electrical noise.