pepe2k/u-boot_mod #98: fsdata.c:333:1: error: expected expression before ',' token

stopspazzing, 2016-04-02:
debian jessie 64bit
Also requires Java installed, which isn't listed in requirements in readme.

>spazz@debian:~/u-boot_mod$ make tplink_wdr3600_43x0
Makefile:154: warning: overriding recipe for target '.PHONY'
Makefile:152: warning: ignoring old recipe for target '.PHONY'
Makefile:240: warning: overriding recipe for target 'board/ar7240/db12x/libdb12x.a'
Makefile:208: warning: ignoring old recipe for target 'board/ar7240/db12x/libdb12x.a'
Makefile:240: warning: overriding recipe for target 'cpu/mips/libmips.a'
Makefile:208: warning: ignoring old recipe for target 'cpu/mips/libmips.a'
Makefile:240: warning: overriding recipe for target 'cpu/mips/ar7240/libar7240.a'
Makefile:208: warning: ignoring old recipe for target 'cpu/mips/ar7240/libar7240.a'
> Configuring for TP-Link TL-WDR3600/43x0 at: Sat Apr 2 16:01:40 PDT 2016 
Makefile:154: warning: overriding recipe for target '.PHONY'
Makefile:152: warning: ignoring old recipe for target '.PHONY'
Makefile:240: warning: overriding recipe for target 'board/ar7240/db12x/libdb12x.a'
Makefile:208: warning: ignoring old recipe for target 'board/ar7240/db12x/libdb12x.a'
Makefile:240: warning: overriding recipe for target 'cpu/mips/libmips.a'
Makefile:208: warning: ignoring old recipe for target 'cpu/mips/libmips.a'
Makefile:240: warning: overriding recipe for target 'cpu/mips/ar7240/libar7240.a'
Makefile:208: warning: ignoring old recipe for target 'cpu/mips/ar7240/libar7240.a'
r - crc32.o
r - ctype.o
r - display_options.o
r - ldiv.o
r - string.o
r - vsprintf.o
r - LzmaWrapper.o
r - LzmaDecode.o
r - main.o
r - cmd_bootm.o
r - cmd_boot.o
r - cmd_flash.o
r - cmd_mem.o
r - cmd_custom.o
r - cmd_net.o
r - cmd_nvedit.o
r - cmd_itest.o
r - cmd_load.o
r - cmd_qcaclk.o
r - command.o
r - console.o
r - devices.o
r - dlmalloc.o
r - environment.o
r - env_common.o
r - env_nowhere.o
r - exports.o
r - flash.o
r - lists.o
r - env_flash.o
r - hush.o
r - xyzModem.o
r - crc16.o
r - s_record.o
r - db12x.o
r - ../common/ar7240_pci.o
r - ../common/spi_flash.o
r - ../common/common.o
r - ../common/athrs17_phy.o
r - cpu.o
r - cache.o
r - qca_common.o
r - qca_clocks.o
r - qca_sf.o
r - qca_dram.o
r - qca_ls_uart.o
r - ag934x.o
r - qca_gpio_init.o
r - qca95xx_pll_init.o
r - board.o
r - time.o
r - mips_linux.o
r - netconsole.o
r - pci.o
r - net.o
r - tftp.o
r - rarp.o
r - eth.o
r - bootp.o
r - sntp.o
r - httpd.o
r - date.o
In file included from fs.c:54:0:
fsdata.c:333:1: error: expected expression before ',' token
 , 0 };
 ^
/home/spazz/u-boot_mod/u-boot/config.mk:248: recipe for target 'fs.o' failed
make[2]: *** [fs.o] Error 1
Makefile:208: recipe for target 'httpd/libhttpd.a' failed
make[1]: *** [httpd/libhttpd.a] Error 2
Makefile:117: recipe for target 'tplink_wdr3600_43x0' failed
make: *** [tplink_wdr3600_43x0] Error 2

pepe2k, 2016-04-03:
Seems to be the same as this one: https://github.com/pepe2k/u-boot_mod/issues/55

jacycle, 2016-04-14:
install java on you pc.
I install openjdk-7-jre, the error disappear.

closed 2016-06-15 by pepe2k