pepe2k/u-boot_mod #100: wr740n_v4 openwrt wlan can't work

jacycle, 2016-04-14:
hello  pepe2k,
  I have a board tplink-wr740n,
  I update the fireware openwrt-ar71xx-generic-tl-wr740n-v4-squashfs-sysupgrade.bin
  when I build u-boot_mod and update, the logs are the following content.
***************************************
*     U-Boot 1.1.4--clean     *
*        Apr 13 2016, 20:45:07        *
***************************************

  BOARD: TP-Link TL-WR74xN/D v4
    SOC: AR9330 rev. 1
    CPU: MIPS 24Kc
    RAM: 64 MB DDR2 16-bit CL3-4-4-10
  FLASH: 16 MB Winbond W25Q128
    MAC: 00:03:7F:09:0B:AD (fixed)
 CLOCKS: CPU/RAM/AHB/SPI/REF
         400/400/200/ 25/ 25 MHz

Hit any key to stop autobooting:  0 

Booting image at: 0x9F020000

   Image name:   OpenWrt r42625
   Image type:   MIPS Linux Kernel Image (lzma compressed)
   Data size:    1107084 Bytes = 1.1 MB
   Load address: 0x80060000
   Entry point:  0x80060000

Uncompressing kernel image... OK!
Starting kernel...
..
..
..
procd: - init complete -
[   24.290000] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   24.290000] device eth0 entered promiscuous mode
[   24.300000] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   24.320000] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   25.150000] eth0: link up (1000Mbps/Full duplex)
[   25.150000] br-lan: port 1(eth0) entered forwarding state
[   25.160000] br-lan: port 1(eth0) entered forwarding state
[   25.160000] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   25.190000] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   27.160000] br-lan: port 1(eth0) entered forwarding state

I use another boot, the logs are the following content.
*********************************************
*        U-Boot 1.1.4  (Jun 19 2014)        *
*********************************************

AP121 (AR9331) U-Boot

DRAM:   64 MB DDR2 16-bit
FLASH:  Winbond W25Q128 (16 MB)
CLOCKS: 400/400/200/25 MHz (CPU/RAM/AHB/SPI)

LED on during eth initialization...

Hit any key to stop autobooting:  0 

Booting image at: 0x9F020000

   Image name:   OpenWrt r42625
   Image type:   MIPS Linux Kernel Image (lzma compressed)
   Data size:    1107084 Bytes = 1.1 MB
   Load address: 0x80060000
   Entry point:  0x80060000

Uncompressing kernel image... OK!
Starting kernel...
..
..
..
procd: - init complete -
[   23.340000] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   23.340000] device eth0 entered promiscuous mode
[   23.350000] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   23.370000] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   24.180000] eth0: link up (1000Mbps/Full duplex)
[   24.180000] br-lan: port 1(eth0) entered forwarding state
[   24.190000] br-lan: port 1(eth0) entered forwarding state
[   24.190000] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   24.230000] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   25.100000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   25.120000] device wlan0 entered promiscuous mode
[   25.130000] br-lan: port 2(wlan0) entered forwarding state
[   25.130000] br-lan: port 2(wlan0) entered forwarding state
[   25.140000] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   26.190000] br-lan: port 1(eth0) entered forwarding state
[   27.130000] br-lan: port 2(wlan0) entered forwarding state

The question is
1. when I use u-boot_mod image, the wifi can't work, use the other boot, it works well.
    login the web, if channged the wifi configuration or led configuration will let the watchdog reset.
2. The chip is ar9331, why the boot print SOC: AR9330 rev. 1

I don't down how to reslove the problem. 
Someone can give me a hand.

pepe2k, 2016-04-14:
Hello @jacycle,

> 2. The chip is ar9331, why the boot print SOC: AR9330 rev. 1

AR9331 == AR9330 rev. 1.

> 1. when I use u-boot_mod image, the wifi can't work, use the other boot, it works well.
> login the web, if channged the wifi configuration or led configuration will let the watchdog reset.

First of all, I see at least two problems with your board:

1. It seems that your MAC address is missing (MAC: 00:03:7F:09:0B:AD (fixed)). Did you combine old content of the second 64 KB sector with new version of my U-Boot modification image?
2. The U-Boot sees DDR2 memory type on your board (RAM: 64 MB DDR2 16-bit CL3-4-4-10, DRAM: 64 MB DDR2 16-bit) which is definitely wrong as the WR740N support only DDR1. That could be a hardware problem - memory type is recognized based on values of GPIO28 and GPIO12.

To be able to find the reason, I need more information from you:

1. What exactly version of the OpenWrt you are using - is it official release or your custom build?
2. Please, post whole two bootlogs.
3. Please, issue in U-Boot command line ``md.l 0x180600ac 1`` and post here output.

Piotr

jacycle, 2016-04-15:
hello pepe2k,
  I changed the spi flash to W25Q128FV, and I changed the ddr. My board is sendcond hand, The name of the board my be mistaken.
  As you said the mac address is missing, I write an address on 0x1fc00, the wifi works.
but sometimes it will be reset by watchdog,

1. the Openwrt is official release
2. uboot> md.l 0x180600ac 1
180600AC: 0004220E    ..".
3. the following are the logs

***************************************
*     U-Boot 1.1.4--clean     *
*        Apr 15 2016, 11:19:14        *
***************************************

** Warning: bad env CRC, using default,
   use 'saveenv' to save it in FLASH

  BOARD: TP-Link TL-WR74xN/D v4
Get revision ID value=111
    SOC: AR9330 rev. 1
    CPU: MIPS 24Kc
    RAM: 64 MB DDR2 16-bit CL3-4-4-10
  FLASH: 16 MB Winbond W25Q128
    MAC: 38:C7:0A:97:0A:D7
 CLOCKS: CPU/RAM/AHB/SPI/REF
         400/400/200/ 25/ 25 MHz

## Error: MDIO_BUSY!
Hit any key to stop autobooting:  0 

Booting image at: 0x9F020000

   Image name:   OpenWrt r42625
   Image type:   MIPS Linux Kernel Image (lzma compressed)
   Data size:    1107084 Bytes = 1.1 MB
   Load address: 0x80060000
   Entry point:  0x80060000

Uncompressing kernel image... OK!
Starting kernel...

[    0.000000] Linux version 3.10.49 (bb@builder1) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 r42625) ) #3 Wed Oct 1 14:00:51 CEST 2014
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 00019374 (MIPS 24Kc)
[    0.000000] SoC: Atheros AR9330 rev 1
[    0.000000] Clocks: CPU:400.000MHz, DDR:400.000MHz, AHB:200.000MHz, Ref:25.000MHz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x03ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x03ffffff]
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line:  board=TL-WR741ND-v4 console=ttyATH0,115200 rootfstype=squashfs,jffs2 noinitrd
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 61076k/65536k available (2369k kernel code, 4460k reserved, 621k data, 272k init, 0k highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:51
[    0.000000] Calibrating delay loop... 265.42 BogoMIPS (lpj=1327104)
[    0.080000] pid_max: default: 32768 minimum: 301
[    0.080000] Mount-cache hash table entries: 512
[... 134 lines cut ...]
** Warning: bad env CRC, using default,
   use 'saveenv' to save it in FLASH

  BOARD: TP-Link TL-WR74xN/D v4
Get revision ID value=111
    SOC: AR9330 rev. 1
    CPU: MIPS 24Kc
    RAM: 64 MB DDR2 16-bit CL3-4-4-10
  FLASH: 16 MB Winbond W25Q128
    MAC: 38:C7:0A:97:0A:D7
 CLOCKS: CPU/RAM/AHB/SPI/REF
         400/400/200/ 25/ 25 MHz

## Error: MDIO_BUSY!
Hit any key to stop autobooting:  0 

Booting image at: 0x9F020000

   Image name:   OpenWrt r42625
   Image type:   MIPS Linux Kernel Image (lzma compressed)
   Data size:    1107084 Bytes = 1.1 MB
   Load address: 0x80060000
   Entry point:  0x80060000

Uncompressing kernel image...

pepe2k, 2016-04-15:
Hello @jacycle,

First of all, your board reports DDR2 memory type and if it's TL-WR74x v4 for sure then you have some hardware problem on the board, with wrong GPIO bootstrap configuration.

Can you take a photo of your board or at least tell me memory chip mark/name? If the memory chip is not in BGA package then it's definitely not DDR2.

Please, try rebuild the code with defined CONFIG_BOARD_DRAM_TYPE_DDR1 (you can add it somewhere here: https://github.com/pepe2k/u-boot_mod/blob/master/u-boot/Makefile#L424). It will ignore bootstrap and setup memory as DDR1. Before you do this, make sure what type of memory you have on the board as it may not boot with DDR1 setup if there is DDR2 on board.

Plus, try to make some memory tests:

1. Under U-Boot: ```mtest```.
2. Under OpenWrt: using memtester (should be available in OpenWrt repository, try ```opkg update; opkg install memtester; memtester 16 3```.

Let me know about the results.

Piotr

jacycle, 2016-04-15:
Hello @pepe2k 
    I try another board, it works well and does not reset. There should be some hardware problems.

   There's a ddr2 in my board, it's in BGA package. 

root@OpenWrt:/# opkg update
Downloading http://downloads.openwrt.org/barrier_breaker/14.07/ar71xx/generic/packages/base/Packages.gz.
Updated list of available packages in /var/opkg-lists/barrier_breaker_base.
Downloading http://downloads.openwrt.org/barrier_breaker/14.07/ar71xx/generic/packages/luci/Packages.gz.
Updated list of available packages in /var/opkg-lists/barrier_breaker_luci.
Downloading http://downloads.openwrt.org/barrier_breaker/14.07/ar71xx/generic/packages/packages/Packages.gz.
Updated list of available packages in /var/opkg-lists/barrier_breaker_packages.
Downloading http://downloads.openwrt.org/barrier_breaker/14.07/ar71xx/generic/packages/routing/Packages.gz.
Updated list of available packages in /var/opkg-lists/barrier_breaker_routing.
Downloading http://downloads.openwrt.org/barrier_breaker/14.07/ar71xx/generic/packages/telephony/Packages.gz.
Updated list of available packages in /var/opkg-lists/barrier_breaker_telephony.
Downloading http://downloads.openwrt.org/barrier_breaker/14.07/ar71xx/generic/packages/management/Packages.gz.
Updated list of available packages in /var/opkg-lists/barrier_breaker_management.
Downloading http://downloads.openwrt.org/barrier_breaker/14.07/ar71xx/generic/packages/oldpackages/Packages.gz.
Updated list of available packages in /var/opkg-lists/barrier_breaker_oldpackages.

thanks for your help.

pepe2k, 2016-04-15:
Hello @jacycle,

It would be good to find the real reason of your problems. Please, make memory tests.
The logs you showed is only about packages list update.

Piotr

jacycle, 2016-04-15:
The new board, when I power down and power up, it will stop to print in somewhere at random and will make watchdog reset. If I enter the system, then for a while it will reset too.
As before I said it works well, I first use the old boot and then update the u-boot_mod,  

I maked some memory tests:

uboot> mtest
Testing RAM 0x80200000...0x83DFFFFF:
Iteration:    325

root@OpenWrt:/# memtester 16 3
memtester version 4.1.3 (32-bit)
Copyright (C) 2010 Charles Cazabon.
Licensed under the GNU General Public License version 2 (only).

pagesize is 4096
pagesizemask is 0xfffff000
want 16MB (16777216 bytes)
got  16MB (16777216 bytes), trying mlock ...locked.
Loop 1/3:
  Stuck Address       : ok         
  Random Value        : ok
  Compare XOR         : ok
  Compare SUB         : ok
  Compare MUL         : ok
  Compare DIV         : ok
  Compare OR          : ok
  Compare AND         : ok
  Sequential Increment: ok
  Solid Bits          : ok         
  Block Sequential    : ok         
  Checkerboard        : ok         
  Bit Spread          : ok         
  Bit Flip            : ok         
  Walking Ones        : ok         
  Walking Zeroes      : ok         

Loop 2/3:
  Stuck Address       : ok         
  Random Value        : ok
  Compare XOR         : ok
  Compare SUB         : ok
  Compare MUL         : ok
  Compare DIV         : ok
  Compare OR          : ok
  Compare AND         : ok
  Sequential Increment: ok

I'm sure the board gpios are as same as tplink_wr740n_v4 the reset gpio and the leds  are working correctly when i use openwrt-ar71xx-generic-tl-wr740n-v4-squashfs-sysupgrade.bin. 

Where is the code I can change the GPIO bootstrap configuration.

pepe2k, 2016-04-15:
Hello @jacycle,

> The new board, when I power down and power up, it will stop to print in somewhere at random and will make watchdog reset. If I enter the system, then for a while it will reset too.
> As before I said it works well, I first use the old boot and then update the u-boot_mod,

I will take a look on this problem. As I see, you are using OpenWrt BB version (kernel 3.10), am I right? Have you tried OpenWrt CC?

> Where is the code I can change the GPIO bootstrap configuration.

If your board has DDR2 memory then your bootstrap configuration is correct. And it doesn't come from code - bootstrap configuration is read from some GPIO values at power up.

Piotr

jacycle, 2016-04-15:
There is a chip of SiGe 2565T for the wifi. 
May be it's the problem.

pepe2k, 2016-04-15:
> There is a chip of SiGe 2565T for the wifi.
> May be it's the problem.

I don't think so, this is just a PA.

jacycle, 2016-04-15:
Hello @pepe2k,

There is a switch for wifi tx and rx
There is a picture here  http://pan.baidu.com/s/1eRSs6Ym

jacycle, 2016-04-16:
Hello @pepe2k,

ar9331 A60(ANTC)  A59(ANTB) to control the wifi tx and rx.
Do I need to initialize in boot?

jacycle, 2016-04-18:
Hello @pepe2k,

there is a signal for low to high and high to low in A60(ANTC) A59(ANTB).
But u-boot_mod, always low level.
Do you know how to configure the registers?

pepe2k, 2016-04-18:
Hello @jacycle,

> ar9331 A60(ANTC) A59(ANTB) to control the wifi tx and rx.
> Do I need to initialize in boot?

> there is a signal for low to high and high to low in A60(ANTC) A59(ANTB).
> But u-boot_mod, always low level.
> Do you know how to configure the registers?

Basically, U-Boot doesn't touch WiFi at all (excluding reset it).

All WiFi related configuration should be done in the system/driver. Your board is custom one, so don't expect that regular OpenWrt image dedicated for TL-WR74xN v4 will work there and configure any external LNA/PA correctly.

IMHO, you should ask manufacturer how to configure external PA. It might be connected to some GPIO.

Piotr

jacycle, 2016-04-18:
Hello @pepe2k,

   I just change the uboot, It works. the same firmware.
   I just want to know what is done in uboot, or what is the different.

 Thank you anyway

pepe2k, 2016-04-18:
Hello @jacycle,

> I just change the uboot, It works. the same firmware.
> I just want to know what is done in uboot, or what is the different.

Changed to which one? And again, have you tried OpenWrt CC version?

Piotr

jacycle, 2016-04-18:
Hello @pepe2k,
    I get the uboot from the board.  (dd if=/dev/mtd0 of=/tmp/u-boot.bin).
    Then I changed the boot to u-boot_mod, after that, ANTB, ANTC is low level always, and someties wil trigger the watchdog.
    If I change back, There is high and low level in the pin of  ANTB ANTC.
    I just changed the boot.
   Do you have the download address for OpenWrt CC version.

pepe2k, 2016-04-18:
Hello @jacycle,

> I get the uboot from the board. (dd if=/dev/mtd0 of=/tmp/u-boot.bin).

And how can I know what is done or not done in stock U-Boot code? I'm not a wizard...
You haven't provided enough information about your board, ex. full stock bootlog.

> Then I changed the boot to u-boot_mod, after that, ANTB, ANTC is low level always, and someties wil trigger the watchdog.

Is the only new code of my project not working on your board?

If you really want help and solve the problem, start to be more verbose please. "u-boot_mod" doesn't tell anything about the version, so next time write please something like "with last release 'Jun 19 2014' it's working, with new code from master branch, compiled using XYZ toolchain, it doesn't" etc.

> Do you have the download address for OpenWrt CC version.

https://downloads.openwrt.org/chaos_calmer/15.05.1/

Piotr

jacycle, 2016-04-18:
Hello @pepe2k,

  In the boot, I input
uboot>setclk 6
Selected profile 6:
- CPU: 200 MHz
- RAM: 200 MHz
- AHB: 200 MHz
- SPI:  25 MHz

after reset, all works fine.
I'm sorry about ANTB and ANTC, they are controlled by openwrt.

pepe2k, 2016-04-18:
Hello @jacycle,

You are not answering my questions, so EOT from me.

Piotr

closed 2016-04-18 by pepe2k

jacycle, 2016-04-18:
Hello @pepe2k,

I forget to say that I also tried OpenWrt CC version, it’s as same as Barrier Breaker 14.07.
CROSS_COMPILE=mips-openwrt-linux-uclibc-
I'm sorry I didn't offer enough verbose. I'll try to do better next time.

thank you for you code, and thank you for the help.