pepe2k/u-boot_mod #101: zsun-wifi-card-reader support?

Lennong, 2016-04-24:
Hello pepe2k! Is it possible to add support for the Zsun WiFi SD Card Reader? It's AR9331 based and a really nice candidate for a weather station project I'm pondering. ;) I would love to see some serious underclocking capabilities if possible. From what I have read you can go so low as 100/100/100 on it. It should give quite some savings I think.

OpenWRT is already supported and flashing is fairly easy. It got really good specs for both the size and cost:
[url]https://wiki.hackerspace.pl/projects:zsun-wifi-card-reader#hardware[/url]

pepe2k, 2016-04-24:
Sure, I will do it, I have this device.

Cheers

Lennong, 2016-04-24:
Great stuff! You the man! :)

Cheers

maurerr, 2016-04-26:
there is already some work done by https://zsunwifi.wordpress.com/u-boot/ but it's 256kb like carambola2 not 64kb as we need

mbee47, 2017-03-15:
Hi, Pepe2k

I looked in your repo but could not find a commit where the zsun wifi reader is supported.
Is this work started?

I also read on lede forum that you have a patch to support this device, do you have a repo with the changes needed to support this device?

regards,

mbee47, 2017-03-16:
Hi @pepe2k 
Thanks for answering.

The question was primary regarding uboot support for this device.
It would be really nice to have an updated uboot. 

Also thanks for the tip about LEDE, Yes I read your comments I just thought you had your own repo with your changes.

regards,

radiolee, 2018-02-04:
u-boot mod still no offically support zsun wifi now?

HenrikNilsson, 2018-03-07:
Afaik not officially, but some work is being done. https://forum.lede-project.org/t/supporting-zsun-wifi-card-reader-16mb-flash-64mb-ram-ar9331/2142/98
Also it helps with commits like: https://github.com/pepe2k/u-boot_mod/commit/2291c2c42eb29cb987637a2322aca8341d9e80a5

Lets hope everything works out and people find time to get this fixed.

closed 2022-10-13 by pepe2k