pepe2k/u-boot_mod #107: How to compile a uboot-unlocked FW for DIR-505？

467815891a, 2016-06-07:
First of all，I really appreciate for your work!I know you have released firmwares with U-boot unlocked to help us flash U-boot easily.But I really want to compile my own uboot-unlocked FW for DIR-505.I refer to some posts and located /target/linux/ar71xx/files/driver/mtd/tplinkpart.c,delete "parts[0].mask_flags = MTD_WRITEABLE;".However,it seems doesn't work for dir-505.Am I change the wrong file?
Besides,I am still learning English,so,forgive my poor English😁

pepe2k, 2016-06-07:
Hello,

For DIR-505, you should change this line:
https://github.com/lede-project/source/blob/master/target/linux/ar71xx/image/legacy.mk#L266

Or better, use this kernel module:
https://github.com/jclehner/mtd-rw

Cheers

467815891a, 2016-06-07:
So I just change "64k(u-boot)ro" to "64k(u-boot)rw" is OK?

pepe2k, 2016-06-07:
> So I just change "64k(u-boot)ro" to "64k(u-boot)rw" is OK?

Just drop "ro" label.

Cheers

closed 2016-06-07 by pepe2k