pepe2k/u-boot_mod #110: Bootloader for LigoWave NFT 1Ni device - AR9341, 64M RAM, 16M (8M) FLASH

ZigFisher, 2016-06-16:
Hi !
I start porting OpenWRT to LigoWave NFT 1Ni device.
After completion of the work, I lay in his GitHub.

Original LigoWave u-boot have dual-boot mode:
dev: size erasesize name
mtd0: 00030000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00120000 00010000 "kernel1"
mtd3: 00660000 00010000 "rootfs1"
mtd4: 00120000 00010000 "kernel2"
mtd5: 00660000 00010000 "rootfs2"
mtd6: 00090000 00010000 "data"
mtd7: 00020000 00010000 "cfg"
mtd8: 00010000 00010000 "art"
mtd9: 00780000 00010000 "firmware1"
mtd10: 00780000 00010000 "firmware2"
mtd11: 01000000 00010000 "flash"

And I need normal single-boot bootloader for AR9341 with 64M RAM and 16M FLASH (second variant - 8M FLASH).

You can help me ?  TNX !

pepe2k, 2016-06-16:
Hello @ZigFisher,

What exactly do you want to know?

Cheers,
Piotr

ZigFisher, 2016-06-20:
I apologize for the delay in replying.
I want to have a bootloader (binary file) for AR9341 with 16M Flash (25q128) and 64M RAM.
I do not have enough knowledge to compile from the source code.
Is it possible to get such a binary file?
TNX !

pepe2k, 2016-06-21:
Hello @ZigFisher,

I haven't tested my code on any real platform with AR9341/AR9342 + DDR2, but I know from one of the users that there is some bug in DDR initialization and the board just doesn't start.

Till I get some AR9341/2 + DDR2 router, test it and eventually solve the problem, I can't help you.

Cheers,
Piotr

pepe2k, 2016-09-25:
Hello @ZigFisher,

DDR2 ssetup on AR9341/2 is now fixed (https://github.com/pepe2k/u-boot_mod/commit/6bcd876a2b13dc9c6db90e15f91754e6c3c2d6d9), I can confirm that it works.

Cheers,
Piotr

closed 2016-09-25 by pepe2k

ZigFisher, 2016-09-27:
Hi !

I test it in next week and send report. Many Thank's !

ZigFisher, 2016-12-22:
Yes, now correct work !
Thank you !