pepe2k/u-boot_mod #111: web doesn't work

yugenchao, 2016-07-14:
Hi pepe2k
     I  compiled TP-Link TL-WR820N for qca9531, uboot can work in addition to web server can not be opened, as if not receive ARP broadcast packets.

pepe2k, 2016-07-20:
I can confirm this problem, please don't close this issue.

yugenchao, 2016-07-20:
ok ,thank you

sven3, 2016-08-13:
I can confirm this on TL-WR703N, using "u-boot-upgrade" from "U-Boot_1.1.4_modification_for_routers_2014-11-19/OpenWrt Barrier Breaker 14.07 images for easy U-Boot upgrade/TP-Link TL-WR703N/openwrt-ar71xx-generic-tl-wr703n-v1-squashfs-sysupgrade.bin".

Pinging the WR703N works, trying to access 192.168.1.1 port 81 (or any other port) results in "connection refused", but accessing 192.168.1.1 port 80 simply gives no response, browser then runs into timeout.

pepe2k, 2016-08-13:
@sven3 
AR933x devices are not affected.

sven3, 2016-08-13:
@pepe2k 
Thanks for the hint. Did I do something wrong then? Should I open a new issue?

pepe2k, 2016-08-13:
@sven3
You can't get "connection refused" using my code... so, you are doing something wrong, probably not even starting web recovery mode.

sven3, 2016-08-15:
@pepe2k 
You are absolutely right... I have no idea what device responded in my LAN, but after connecting my PC to the WR703N directly everything works fine. Sorry for the noise and thanks for the great mod!

commit c879e76e (2016-08-31): Cleanups in QCA953x eth/phy related code

Get rid of some warnings from compilator.
This also fixes #111 (don't drop broadcast packets).
closed 2016-08-19 by pepe2k