pepe2k/u-boot_mod #112: i need i2c-gpio

gandol2, 2016-07-19:
Hello pepe 2k.
I would like to use the i2c-gpio.
I am a novice in the u-boot.
Please help, I might add i2c-gpio driver.
The well-organized web pages that you can add drivers required.

pepe2k, 2016-07-19:
Hello @gandol2,

Why do you need i2c in U-Boot for?

Cheers!

gandol2, 2016-07-19:
@pepe2k 

I'm going to use i2c base tpm(http://www.infineon.com/cms/en/product/security-and-smart-card-solutions/optiga-embedded-security-solutions/optiga-tpm/SLB+9645+XT12+FW13332/productType.html?productType=db3a30433fcce646013fd263b91724a6).

ar9331 does not support i2c. but i need i2c. 

So I'm going to use to fix the u-boot mod.

i will port shousoft_i2c.c. (https://github.com/lentinj/u-boot/blob/master/drivers/i2c/soft_i2c.c)

but i don't know how to port.. plz help me..

pepe2k, 2016-07-19:
@gandol2

It's not a fix, it's more like an additional functionality and it's up to you to make it working.
For now I don't see any reason to include it here.

Mainline U-Boot got basic support for AR933x several months ago.
IMHO, it would be better for you to use it instead of this modification.

Cheers!

gandol2, 2016-07-19:
To me mainline U-Boot is hard...

many source code and config, but u-boot_mod is simple. 

I feel sorry. Thank you..

closed 2016-07-19 by pepe2k