pepe2k/u-boot_mod #116: More Overclock step for AR934x soc

Ansuel, 2016-08-11:
Why does the step stop to 650mhz?
My wdr3600 can go to 780 mhz so why don't increase that? 
(or if you can explain me how to add more, i looked the source but i can't undestand how to add them
(i know i need to edit this qca95xx_pll_init.h)

pepe2k, 2016-08-15:
Hello @Ansuel,

> Why does the step stop to 650mhz?

One reason is safety, but I just haven't got yet time to test more profiles ;)

> My wdr3600 can go to 780 mhz so why don't increase that?
> (or if you can explain me how to add more, i looked the source but i can't undestand how to add them
> (i know i need to edit this qca95xx_pll_init.h)

Every profile setups several registers which are later used to configure PLL/s and clocks during low level initialization.

You can find all needed information about PLL/clock configuration in AR9344 datasheet... available for example here: https://github.com/Deoptim/atheros

Cheers

stachi, 2016-11-30:
I think you can calculate the CPU clock by this formula
```
cpu_mhz  = _nint * XTAL_MHZ / _refdiv;
cpu_mhz += _nfracmin * XTAL_MHZ / (_refdiv * (1 << 6));
cpu_mhz /= (1 << _outdiv);
```

So you can get you the right values for the functions
```
_qca95xx_cpu_pll_cfg_reg_val
_qca95xx_cpu_pll_dither_reg_val
```

But I don't know what effect _range has.
It should be the VCO frequency range of the CPU PLL.
The only info I get is, it shouldn't impact clock values.
But on some configurations you set it to 1.
I dont't know if it is important.

pepe2k, 2016-11-30:
@stachi take a look at page 139 in AR9344 datasheet (link in my reply above), there is information about RANGE field and its value.

Anyway, I'm going to rebuild (again...) PLL and clocking part soon.

Cheers

stachi, 2016-11-30:
@pepe2k thanks for the info.

I now know what it does, but I still can't say when it's important.
I would say that some cpu clocks just run better with different VCO ranges?

But I know nothing about CPU/PLL configuration :)
Just trying to learn something.

pepe2k, 2016-11-30:
> I now know what it does, but I still can't say when it's important.
> I would say that some cpu clocks just run better with different VCO ranges?

QC/A WiSOCs are not designed by me, so all of this are just my assumptions... ;)

It seems that built-in VCO (voltage controlled oscillators) have different (two) frequency operating ranges. The RANGE value should be adjusted according to the VCO frequency output. As both ranges overlap, I assume that important are mostly upper and lower values of the VCO outputs.

Cheers

stachi, 2016-11-30:
@pepe2k i forgot the divider in _qca95xx_cpu_ddr_clk_ctrl_reg_val
Now i see what you mean.

stachi, 2016-12-03:
Based on the formula and the datasheet I added 2 profiles for AR934x soc (wdr3600 in my case).
700_450_225_28 works fine on my router (same clocks I used with the old u-boot mod).
With 750_450_225_28 the router didn't boot anymore.
The PLL configuration should be right. My router just don't like too much OC.

```
{
700, 450, 225, 28,
_qca95xx_spi_ctrl_addr_reg_val(8, 1, 0, 2),
{
	_qca95xx_cpu_pll_cfg_reg_val(28, 1, 0, 0, 0),
	_qca95xx_ddr_pll_cfg_reg_val(18, 1, 1, 0, 0),
	_qca95xx_cpu_ddr_clk_ctrl_reg_val(1, 1, 2, 1, 1, 1),
	_qca95xx_cpu_pll_dither_reg_val(0),
	_qca95xx_ddr_pll_dither_reg_val(0)
}, {
	_qca95xx_cpu_pll_cfg_reg_val(17, 1, 0, 0, 0),
	_qca95xx_ddr_pll_cfg_reg_val(11, 1, 1, 0, 0),
	_qca95xx_cpu_ddr_clk_ctrl_reg_val(1, 1, 2, 1, 1, 1),
	_qca95xx_cpu_pll_dither_reg_val(32),
	_qca95xx_ddr_pll_dither_reg_val(256)
},
}, {
750, 450, 225, 28,
_qca95xx_spi_ctrl_addr_reg_val(8, 1, 0, 2),
{
	_qca95xx_cpu_pll_cfg_reg_val(30, 1, 0, 0, 0),
	_qca95xx_ddr_pll_cfg_reg_val(18, 1, 1, 0, 0),
	_qca95xx_cpu_ddr_clk_ctrl_reg_val(1, 1, 2, 1, 1, 1),
	_qca95xx_cpu_pll_dither_reg_val(0),
	_qca95xx_ddr_pll_dither_reg_val(0)
}, {
	_qca95xx_cpu_pll_cfg_reg_val(18, 1, 0, 0, 0),
	_qca95xx_ddr_pll_cfg_reg_val(11, 1, 1, 0, 0),
	_qca95xx_cpu_ddr_clk_ctrl_reg_val(1, 1, 2, 1, 1, 1),
	_qca95xx_cpu_pll_dither_reg_val(48),
	_qca95xx_ddr_pll_dither_reg_val(256)
},
```

```
[    0.000000] Memory: 125008K/131072K available (3070K kernel code, 158K rwdata, 768K rodata, 284K init, 205K bss, 6064K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:51
[    0.000000] Clocks: CPU:700.000MHz, DDR:450.000MHz, AHB:225.000MHz, Ref:40.000MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5460744132 ns
[    0.000007] sched_clock: 32 bits at 350MHz, resolution 2ns, wraps every 6135667710ns
[    0.008685] Calibrating delay loop... 348.97 BogoMIPS (lpj=1744896)
[    0.071725] pid_max: default: 32768 minimum: 301
[    0.076980] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.084363] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.094548] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.106623] NET: Registered protocol family 16
[    0.112834] MIPS: machine is TP-LINK TL-WDR3600/4300/4310
```

Ansuel, 2016-12-03:
Pls add this

stachi, 2016-12-03:
@Ansuel  You can add oc profiles by editing this files
```
u-boot/include/cmd_qcaclk.h
u-boot/include/soc/qca95xx_pll_init.h
u-boot/include/soc/qca_pll_list.h
```

ThaiCat, 2016-12-09:
as far as i know only only cmd_qcaclk.h has any effect on overclocking profiles

closed 2017-12-08 by pepe2k