pepe2k/u-boot_mod #124: How to use it?

sudden6, 2016-10-14:
Probably a stupid question, but I haven't found any information about how to get to the download page of your modified u-boot, can you point me to it or mention it in the README?

pepe2k, 2016-10-15:
https://github.com/pepe2k/u-boot_mod#ready-snapshottest-images

Cheers,
Piotr

sudden6, 2016-10-15:
I just realized this was worded badly, I meant the instructions how to get to the webinterface of your bootloader. ie do I have to press any special buttons, load at the right time, or something else?

pepe2k, 2016-10-17:
If you have serial access, you can interrupt booting and issue "httpd" command which will start web server. Or, use "reset" button, as described here: https://github.com/pepe2k/u-boot_mod#other

Cheers,
Piotr

sudden6, 2016-10-17:
Thank you very much, IMHO this could be moved https://github.com/pepe2k/u-boot_mod#how-to-use-it so it's easier to find :)

closed 2016-10-17 by sudden6