pepe2k/u-boot_mod #127: How to build RAM version?

ander-wrt, 2016-11-01:
It looks like building of RAM version is broken.
(at least for tp-link mr3020)

I could build ROM version.
ROM version started successfully. I can see logo:

U-Boot 1.1.4-38d83d2c-dirty

Next I've changed "# IMG_RAM =" to "export IMG_RAM = 1" in Makefile
(doesn't work without "export"),

and used commands for building:

make clean
make tp-link_tl-mr3020

I've got error:
u-boot_mod/u-boot/cpu/mips/ar7240/qca_gpio_init.S:447:5: error: #error "Cannot use JTAG and GPIO6~8 at the same time!"
    #error "Cannot use JTAG and GPIO6~8 at the same time!"
     ^
.../u-boot_mod/u-boot/config.mk:242: recipe for target 'qca_gpio_init.o' failed
make[2]: *** [qca_gpio_init.o] Error 1
Makefile:291: recipe for target 'cpu/mips/ar7240/libar7240.a' failed
make[1]: *** [cpu/mips/ar7240/libar7240.a] Error 2
Makefile:203: recipe for target 'tp-link_tl-mr3020' failed
make: *** [tp-link_tl-mr3020] Error 2

Is there any "how-to" for building and usage RAM version?

pepe2k, 2016-11-02:
Hello @ander-wrt,

It's not a bug, it's a feature! :wink:

RAM version is dedicated for users who want to upload the image using JTAG, so in this version JTAG is enabled by default. On AR933x it means that GPIO 6, 7 and 8 can't be used as regular GPIOs (ex. for LEDs or buttons). MR3020 uses some of JTAG GPIOs, so you should first remove them and then build RAM version image. Please, take a look here: https://github.com/pepe2k/u-boot_mod/blob/master/u-boot/include/configs/ap121.h#L85

I will think how to make it more automatic.

Cheers,
Piotr

ander-wrt, 2016-11-02:
Hi @pepe2k ,

Thanks for fast responding.
Removing GPIO8 in line 89 at [](url)https://github.com/pepe2k/u-boot_mod/blob/master/u-boot/include/configs/ap121.h#L89 helped.
Building finished without errors now.

You're absolutely right. I'm using openOCD + J-Link for downloading RAM version into mr3020 board.
But with latest RAM version I've met the same behaviour as I pointed here: [](url)https://github.com/pepe2k/u-boot_mod/issues/48
RAM version can not start stable from run to run. It crashes at relocation point. But occasionally if not crashes it works very stable. I've tried start mtest for long time. There is no any errors!

It looks like there is something wrong with low_level hardware initialization (via JTAG or some piece of code).

Do you have any idea?

pepe2k, 2016-11-02:
> It looks like there is something wrong with low_level hardware initialization (via JTAG or some piece of code).

> Do you have any idea?

Have you tested some older version of my code? Can you show me your setup code for the SOC?

ander-wrt, 2016-11-02:
> Have you tested some older version of my code?
> Can you show me your setup code for the SOC?

Description of old version I've used is here: [](url)https://github.com/pepe2k/u-boot_mod/issues/48

Setup code for the SOC.
I'm using this one: [](url)https://github.com/Deoptim/atheros/blob/master/AR9331.cfg

ander-wrt, 2016-11-02:
I forgot one more thing:
for running RAM version of latest git code I have to put value 0x12345678 at address 0xBD000000 via JTAG or replace "bne" to "nop" at [](url)https://github.com/pepe2k/u-boot_mod/blob/master/u-boot/cpu/mips/ar7240/qca_gpio_init.S#L492
full_reset happens without that and u-boot doesn't start second time as it should be according to near comments.

pepe2k, 2016-11-03:
> I forgot one more thing: [...]

OK, this is actually a bug, I will fix it.
For JTAG/RAM version related issue, I need more time.

Cheers,
Piotr

ander-wrt, 2016-12-26:
Dear Piotr,

Is there any update for this issue or not yet?

Regards,
Andrey.

pepe2k, 2016-12-29:
Hello @ander-wrt,

Can you run regular FLASH version, check what values are written into QCA_DDR_TAP_CTRL_0_REG (0x1800001c) and QCA_DDR_TAP_CTRL_1_REG (0x18000020), and then use the same in your OpenOCD config?

Cheers

ander-wrt, 2017-01-31:
Hello @pepe2k,

I'm sorry for keeping silent. I was busy a bit.
I've tried your suggestion without success unfortunately :-(

Got from regular FLASH version:

uboot> md.l 0x1800001c 1
1800001C: 0000000E    ....
uboot> md.l 0xb800001c 1
B800001C: 0000000E    ....
uboot> md.l 0x18000020 1
18000020: 0000000E    ....

Used the same values for OpenOCD config:

	mww 0xb800001c 0xE
	mww 0xb8000020 0xE

But RAM version hungs at usual point:

``***************************************``
``*                                     *``
``*``             RAM VERSION             ``*``
``*``                                    ``*``
``***************************************``
``*``     U-Boot 1.1.4-38d83d2c-dirty     ``*``
``*``          Build: 2016-11-01          ``*``
``***************************************``

Top of RAM usable for U-Boot at: 81F00000
Reserving 217k for U-Boot at: 81EC8000
Reserving 516k for malloc() at: 81E47000
Reserving 48 Bytes for Board Info at: 81E46FD0
Reserving 36 Bytes for Global Data at: 81E46FAC
Reserving 512k for boot params() at: 81DC6FAC
Stack Pointer at: 81DC6F88
Call relocate_code addr_sp=81DC6F88, addr=81EC8000

Can you suggest anything else?

Regards,
Andrey.

ander-wrt, 2017-01-31:
I'm sorry. Closed by mistake.

ander-wrt, 2017-01-31:
Dear Piotr,

I've found one more thing:
as I told before, sometime RAM version can start successfully.
In that case I can restart openOCD, reload u-boot RAM binary and restart it
successfully on the fly (without power off). RAM version can restart always OK.
But if RAM version hungs, restarting openOCD failed at "PLL init" point.
I have to power off and try to start RAM version again and again.

I hope it will be useful.

Andrey.

ntzrmtthihu777, 2017-05-05:
Just an idea in regards to this: for some basic testing I found it useful to load the elf version of the ram-only build with tftp and the stock u-boot for testing purposes (mostly to double check the gpios for LEDs and buttons). Could be useful to add the elf build to outputs generated by IMG_RAM.

closed 2022-10-13 by pepe2k