pepe2k/u-boot_mod #135: Chain-load u-boot with the RAM version?

zorun, 2017-01-20:
Hi,

I'd like to chain-load two u-boot on the same device.  Basically, keep the stock u-boot unchanged, and add a (more complete) u-boot image where the kernel is normally stored in flash.

As far as I can see, the second-stage u-boot should omit all the low-level hardware initialisation.  Could the RAM version be used for this purpose?

I.e. if I basically dump your RAM version of u-boot on the kernel partition, will it be booted by the stock u-boot?

Thanks!

pepe2k, 2017-01-26:
Hello @zorun,

> I'd like to chain-load two u-boot on the same device. Basically, keep the stock u-boot unchanged, and add a (more complete) u-boot image where the kernel is normally stored in flash.

Why don't you just swap the stock U-Boot?

> As far as I can see, the second-stage u-boot should omit all the low-level hardware initialisation. Could the RAM version be used for this purpose?

In general, yes - you would need a version without low level initialization.

> I.e. if I basically dump your RAM version of u-boot on the kernel partition, will it be booted by the stock u-boot?

No, it won't work this way.

Loading the kernel is little bit more complicated. I suppose your stock U-Boot uses `bootm` command for that, which expects image with an U-Boot header (made with `mkimage`). If:

- it's possible to change `bootcmd` environment variable in your stock U-Boot and
- `go` command is available

then it should be possible. All you would need to do is place _RAM-loadable_ U-Boot image somewhere in FLASH (lets say at `flash_addr` offset) and change `bootcmd` to something like `cp.b flash_addr ram_load_addr image_size && go ram_load_addr`

Eventually, you can try prepare custom image header for RAM U-Boot version and load it with `bootm`. I haven't tried that.

Cheers,
Piotr

closed 2017-01-26 by pepe2k

zorun, 2017-01-26:
Ok, thanks for all the explanations!  I will try both solutions, because the stock u-boot has not persistent environment partition, so having to change `bootcmd` at each boot is cumbersome.

> Why don't you just swap the stock U-Boot?

Because it's risky (the device will be bricked if something goes wrong), and because I want to enable a lot of u-boot features, so the resulting image will probably not fit in the 128k u-boot partition.

pepe2k, 2017-01-26:
> Because it's risky (the device will be bricked if something goes wrong), and because I want to enable a lot of u-boot features, so the resulting image will probably not fit in the 128k u-boot partition.

Can you tell what device it is?

128 KB is a lot of space, especially for compressed U-Boot image. Take a look how it looks with TP-Link devices:

[image]

zorun, 2017-01-26:
It's a TL-WA750RE.  The reference design from qualcomm is called "ap123".  Some models that are supported by your mod look very similar, I plan to have a closer look soon.

For the size issue, I was trying with the latest u-boot, and the binary images are just too big...  Upstream u-boot doesn't seem to support compression though, but even manually gzipping the image only brings it down to 130 KB.  The older u-boot 1.1.4 is maybe more lightweight then :)

pepe2k, 2017-01-26:
> It's a TL-WA750RE. The reference design from qualcomm is called "ap123". Some models that are supported by your mod look very similar, I plan to have a closer look soon.

So it's AR9341 based, there are some boards with this SOC already supported here. You can try... but, as I can see, this one is inside that ugly, glued/welded enclosure so if something goes wrong, you will have nice paperweight ;)

> For the size issue, I was trying with the latest u-boot, and the binary images are just too big... Upstream u-boot doesn't seem to support compression though, but even manually gzipping the image only brings it down to 130 KB. The older u-boot 1.1.4 is maybe more lightweight then :)

Yes, mainline U-Boot is much bigger now, with all nice features it needs ~300 KB which for most QC/A based devices on market is too much.

zorun, 2017-01-26:
> I can see, this one is inside that ugly, glued/welded enclosure so if something goes wrong, you will have nice paperweight ;)

Oh, no problem, I already opened it to have access to the serial console ;)

pepe2k, 2017-01-26:
> Oh, no problem, I already opened it to have access to the serial console ;)

If you have Raspberry Pi or some Arduino, you can make SPI NOR FLASH programmer with them, just in case ;)