pepe2k/u-boot_mod #139: Can't lon in by root after flash,need password

jack338c, 2017-02-09:
(Dear pepe2k,I flash the OpenWrt Barrier Breaker 14.07 file  for my dir505,after reboot ,I just can't log in by ssh,it's need password ,what's the matter?)

Using OpenWrt

This method is recommended!

Starting from official release "2014-11-19", you will find ready OpenWrt images, with unlocked u-boot partition, embedded U-Boot image and dedicated small script for easy update process inside release tarball. All you need to do is download last release, select proper OpenWrt image for your device, install it and invoke one command: u-boot-upgrade:

root@OpenWrt:/# u-boot-upgrade

=================================================================
     DISCLAIMER: you are using this script at your own risk!

     The author of U-Boot modification and this script takes
     no responsibility for any of the results of using them.

          Updating U-Boot is a very dangerous operation
        and may damage your device! You have been warned!
=================================================================
   Are you sure you want to continue (type 'yes' or 'no')? yes
=================================================================

[ ok ] Found U-Boot image file: uboot_for_tp-link_tl-mr3020.bin
       Do you want to use this file (type 'yes' or 'no')? yes
[ ok ] MD5 checksum of new U-Boot image file is correct
[ ok ] Backup of /dev/mtd0 successfully created
       Do you want to store backup in /etc/u-boot_mod/backup/ (recommended, type 'yes' or 'no')? no
[ ok ] New U-Boot image successfully combined with backup file
[info] New U-Boot image is ready to be written into FLASH
       Are you sure you want to continue (type 'yes' or 'no')? yes
[ ok ] New U-Boot image successfully written info FLASH
[ ok ] MD5 checksum of mtd0 and new U-Boot image are equal
[info] Done!

pepe2k, 2017-02-09:
Your question doesn't seem to be related with this project.
Please, read the OpenWrt Wiki.

closed 2017-02-09 by pepe2k

jack338c, 2017-02-09:
I use telnet,it works,anyway,thank you !