pepe2k/u-boot_mod #142: upgrad new uboot for dir505

jack338c, 2017-02-18:
Dear Pepe,I have upgraded my dir505 uboot by Using OpenWrt from you,thank you!

Now I want upgraded my dir505 uboot to u-boot_mod__d-link_dir-505__20170212__git_master-a077b7ee.bin from your iste.Could you tell me the method?

pepe2k, 2017-03-02:
Hi @jack338c,

Sorry for late reply... which version of my modification did you use on your router, the one included in last release (2014-11-19)?

Cheers,
Piotr

jack338c, 2017-03-03:
Hi Piotr

thanks for your reply ,it's ok,everyone is busy except me 

currentIy I use the version--2014-11-19 for dir505,now I want try the neweset version 20170212,but I did not know how to do it

your help will be appericiated 

thanks again 
Best regard
Jack

jack338c, 2017-06-28:
sorry,I find the answer in the manual
thank you Mr Pepe

closed 2017-06-28 by jack338c