pepe2k/u-boot_mod #143: Please add support for TL-WR842N V3

mile18, 2017-03-02:
Hi there, could you add support for TL-WR842N V3. I'll give you all info that you need, also here is some of it https://wiki.openwrt.org/toh/tp-link/tl-wr842nd
Thank you in advance.
Btw. none of the existing QCA9531 images works.

mile18, 2017-03-02:
I think why it doesnt work is becouse of Gigadevice gd25q128 memory is not recognised in your compiled build, could you add in any of existing QCA9531 builds and I will test it?

pepe2k, 2017-03-02:
Hi @mile18,

> I think why it doesnt work is becouse of Gigadevice gd25q128 memory is not recognised in your compiled build, could you add in any of existing QCA9531 builds and I will test it?

I really doubt that the FLASH chip type is the reason.

> Btw. none of the existing QCA9531 images works.

Provide me please more details: which image did you try, how did you write it and what does it mean "doesn't work" exactly (are you getting something on serial console, etc.)?.

Cheers,
Piotr

mile18, 2017-03-02:
I've tryed all yours QCA9531 images (all router models that you have compiled on your server). Flashed it with serial programmer directly on flash chip, haven't tryed serial console. Router only blink's once and then turn itself off. Thank you on your fast answer.

pepe2k, 2017-03-02:
> I've tryed all yours QCA9531 images (all router models that you have compiled on your server). Flashed it with serial programmer directly on flash chip, haven't tryed serial console. Router only blink's once and then turn itself off. Thank you on your fast answer.

What firmware are you using? I hope you have read README, especially [this point](https://github.com/pepe2k/u-boot_mod#known-issues). Try to get access to the serial line, it will tell you if it's working or not.

Cheers,
Piotr

mile18, 2017-03-02:
I am using LEDE 17.01.0, also know how to us it, have tryed on many routers. I'll get serial to usb and try and report back.

mile18, 2017-04-01:
Hi again, i'm back with some informations. Uboot for Comfast CF-E320N v2 does work, but reset button is mapped wrongly and cant enter uboot menu's, also LED-s are not configured correctly. If you could use 841nd's LED-s and button settings i think it will work.

pepe2k, 2017-05-02:
Hi @mile18,

Sorry for late reply. Please, try [this one](http://projects.dymacz.pl/test/u-boot_mod__tp-link_tl-wr842n_v3__20170502__git_dev-a8b49cea.tar.gz) and let me know if it works for you.

Cheers,
Piotr

mile18, 2017-05-04:
No problem, it works :) thank you once more

commit 5c58e34b (2017-05-04): Add support for TP-Link TL-WR842N/D v3 (QCA9531 based)

This closes #143.
closed 2017-05-04 by pepe2k