pepe2k/u-boot_mod #145: Please add support for QCA956x (TP-LINK TL-WR1043ND v4)

VittGam, 2017-03-23:
Hello,

The new TP-LINK TL-WR1043ND v4 is based on QCA956x SoC (quite different from QCA955x of previous generation) and AR8337 switch chip (99.99% software compatible with AR8327, except for quirks and workarounds...).

It would be great to support it here.

The original u-boot uses TP-LINK "safeloader" new image format, but I personally don't think support for it would be needed here. I would be more than happy with a mktplinkfw (standard) compatible bootloader!

If I can somehow help, please tell me and I will try... :)

Cheers,
Vittorio

ptpt52, 2017-12-16:
need this support too!
tp-link wdr6500 v7 is QCA956x soc

vinibali, 2021-07-31:
AC1750 v4 is based on QCA9563 too.
@pepe2k, are you planning to do anything with the router you got previously?
_Thanks to M-K O'Connell for donating a router with QCA9563_

closed 2022-10-13 by pepe2k