pepe2k/u-boot_mod #148: Make error CFG_ENV_SIZE undeclared.

ntzrmtthihu777, 2017-04-14:
Greetings. I'd like to think I'm not a total idiot, but I can't seem to get this to compile right.
No matter how I go about it, I receive the following error:
```bash
In file included from environment.c:35:0:
../include/environment.h:105:21: error: ‘CFG_ENV_SIZE’ undeclared here (not in a function)
  unsigned char data[ENV_SIZE]; /* Environment data  */
                     ^
make[2]: *** [Makefile:138: environment.o] Error 1
make[1]: *** [Makefile:297: tools] Error 2
make: *** [Makefile:217: tp-link_tl-wr710n] Error 2
```
Mostly this is meant to be a test build so I can begin to try and port to my board (EnGenius/Senao ENS202{,EXT}).

pepe2k, 2017-04-14:
Hi @ntzrmtthihu777,

Unfortunately, I'm not able to read in minds, so try to be more specific.

Cheers,
Piotr

ntzrmtthihu777, 2017-04-14:
Well, that's just it. I add the toolchain's bin dir to $PATH, and run `make tp-link_tl-wr710n`, and I get that error. I'm using the lede toolchain (built it myself, working on porting ens202ext to lede). There's not much to explain.

pepe2k, 2017-04-14:
It works for me:
```
PATH=$PATH:/home/.../staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/bin/
make tp-link_tl-wr710n

[...]

Copying compiled image...
Checking size of the image...
Preparing 123 KB image padded with 0xFF...
Preparing final image...
Calculating MD5 sum for the final image...

DONE!
Image 'bin/u-boot_mod__tp-link_tl-wr710n__20170414__git_master-7e7a2ab4.bin' is ready!
```

ntzrmtthihu777, 2017-04-14:
Simply does not work for me. Same command, export to the same dir for me. Could it be your local copy is different from this repo? Exact same error.

pepe2k, 2017-04-14:
> Could it be your local copy is different from this repo?

No :)

Show me whole output from your shell, from begin to the error.

ntzrmtthihu777, 2017-04-14:
```bash
$ export PATH=$PATH:/home/.../staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/bin/
$ make tp-link_tl-wr710n 
Removing configuration...
Preparing configuration for target: tp-link_tl-wr710n

  Device vendor/model:	TP-Link TL-WR710N
  Custom recovery web:	no
  Image size limit:	123 KB
  Default FLASH size:	8 MB
  GPIO reset button:	11
  Button active low:	no
  LZMA compression:	yes
  RAM-loadable only:	no

Preparing web server files...

In file included from environment.c:35:0:
../include/environment.h:105:21: error: ‘CFG_ENV_SIZE’ undeclared here (not in a function)
  unsigned char data[ENV_SIZE]; /* Environment data  */
                     ^
make[2]: *** [Makefile:138: environment.o] Error 1
make[1]: *** [Makefile:297: tools] Error 2
make: *** [Makefile:215: tp-link_tl-wr710n] Error 2
```

pepe2k, 2017-04-14:
Show me please outputs from (after running make):
`cat u-boot/include/config.h` and `git status`

ntzrmtthihu777, 2017-04-14:
```c
#include <soc/soc_list.h>
#define CONFIG_BUILD_DATE_UTC			2017-04-14
#define CFG_AG7240_NMACS			2
#define CFG_ATHRS26_PHY				1
#define CONFIG_MACH_HORNET			1
#define CONFIG_HOSTNAME				u-boot_tl-wr710n
#define CONFIG_GPIO_RESET_BTN			11
#define CONFIG_MAX_UBOOT_SIZE			125952
#define CONFIG_MAX_UBOOT_SIZE_HEX		0x1EC00
#define SOC_TYPE				QCA_AR933X_SOC
#define CONFIG_BOARD_CUSTOM_STRING		TP-Link TL-WR710N
#define CONFIG_DEFAULT_FLASH_SIZE_IN_MB		8
#define CONFIG_FOR_TPLINK_WR710N_V1		1

/* Automatically generated - do not edit */
#include <configs/ap121.h>
```
```On branch master
$ git status
Your branch is up-to-date with 'origin/master'.
nothing to commit, working tree clean
$ git remote -v
origin	git@github.com:ntzrmtthihu777/u-boot_mod.git (fetch)
origin	git@github.com:ntzrmtthihu777/u-boot_mod.git (push)
```
I have a fork that I'm hoping to port my board with. I've made no changes yet.

pepe2k, 2017-04-14:
OK, that's weird... I suppose something is missing on your system.

Please, remove `--silent --no-print-directory` from this line https://github.com/pepe2k/u-boot_mod/blob/master/Makefile#L32, run `make lzma_host_clean`, `make clean_all`, then again `make tp-link_tl-wr710n` and show me output.

ntzrmtthihu777, 2017-04-14:
```bash
$ make tp-link_tl-wr710n 
make[1]: Entering directory '/home/netz/Projects/u-boot_mod/u-boot'
Removing configuration...
Preparing configuration for target: tp-link_tl-wr710n

  Device vendor/model:	TP-Link TL-WR710N
  Custom recovery web:	no
  Image size limit:	123 KB
  Default FLASH size:	8 MB
  GPIO reset button:	11
  Button active low:	no
  LZMA compression:	yes
  RAM-loadable only:	no

make[1]: Leaving directory '/home/netz/Projects/u-boot_mod/u-boot'
make[1]: Entering directory '/home/netz/Projects/u-boot_mod/u-boot'
echo -e "\e[92mPreparing web server files...\e[0m"
Preparing web server files...

cd httpd && ./vendors/makefsdatac 
make[2]: Entering directory '/home/netz/Projects/u-boot_mod/u-boot/tools'
ln -s ../common/environment.c environment.c
ln -s ../lib_generic/crc32.c crc32.c
make[2]: '.depend' is up to date.
make[2]: Leaving directory '/home/netz/Projects/u-boot_mod/u-boot/tools'
make -C tools all
make[2]: Entering directory '/home/netz/Projects/u-boot_mod/u-boot/tools'
gcc -g -Wall -pedantic -idirafter ../include -DTEXT_BASE=0x80010000 -DUSE_HOSTCC -O2 -c mkimage.c
gcc -g -Wall -pedantic -idirafter ../include -DTEXT_BASE=0x80010000 -DUSE_HOSTCC -O2 -c crc32.c
gcc -Wall -pedantic -idirafter ../include -DTEXT_BASE=0x80010000 -DUSE_HOSTCC -O2  -o mkimage mkimage.o crc32.o
strip mkimage
gcc -g -Wall -pedantic -idirafter ../include -DTEXT_BASE=0x80010000 -DUSE_HOSTCC -O2 -c envcrc.c
gcc -g  -idirafter ../include -DTEXT_BASE=0x80010000 -DUSE_HOSTCC -c environment.c
In file included from environment.c:35:0:
../include/environment.h:105:21: error: ‘CFG_ENV_SIZE’ undeclared here (not in a function)
  unsigned char data[ENV_SIZE]; /* Environment data  */
                     ^
make[2]: *** [Makefile:138: environment.o] Error 1
make[2]: Leaving directory '/home/netz/Projects/u-boot_mod/u-boot/tools'
make[1]: *** [Makefile:297: tools] Error 2
make[1]: Leaving directory '/home/netz/Projects/u-boot_mod/u-boot'
make: *** [Makefile:223: tp-link_tl-wr710n] Error 2```

pepe2k, 2017-04-14:
What system you are building on? Show me also `gcc --version`.

ntzrmtthihu777, 2017-04-14:
Gentoo (fairly up to date, ether yesterday or the day before).
gcc (Gentoo 6.3.0 p1.0) 6.3.0

pepe2k, 2017-04-14:
> gcc (Gentoo 6.3.0 p1.0) 6.3.0

That's probably a reason.
I don't have time now to fix building with gcc 6, can you try with older one?

ntzrmtthihu777, 2017-04-14:
sure. I have 5.4 as well, its just not the default.

ntzrmtthihu777, 2017-04-14:
False hope. after switching to gcc-5.4.0 I still have the same error.

pepe2k, 2017-04-14:
Try `make clean` before new build.

ntzrmtthihu777, 2017-04-14:
No joy, even after running clean_all, clean, lzma_host_clean.

pepe2k, 2017-04-14:
I really don't know why it doesn't work for you.
Try add `#include <config.h>` above `#include <environment.h>` inside u-boot/common/environment.c

ntzrmtthihu777, 2017-04-14:
Seems its already there with the asm macros, but adding it again outside of the macro doesn't fix it.

pepe2k, 2017-04-14:
Short google search showed that you are not the first one with similar problem... something is wrong with your system/environment configuration.

Last try... Does it build if you change `#include <config.h>` to `#include "../include/config.h"` in u-boot/common/environment.c?

ntzrmtthihu777, 2017-04-14:
That did it! Something must be funky there, then... Good enough workaround for now, though.

On another note, what would I need information-wise to start porting for my board?

pepe2k, 2017-04-14:
> That did it! Something must be funky there, then... Good enough workaround for now, though.

No idea what and where is wrong...

> On another note, what would I need information-wise to start porting for my board?

Not much, GPIO numbers for LEDs, reset button, some FLASH offsets (environment, MAC, kernel, art), etc. Look into commit history, for example: https://github.com/pepe2k/u-boot_mod/commit/3fb87f47a54d45b87828b080fb1bfe06ba07c7c8 (if I'm correct, your board is AR9341 based).

ntzrmtthihu777, 2017-04-14:
And how would one check if a given u-boot is lzma'd?

pepe2k, 2017-04-14:
> And how would one check if a given u-boot is lzma'd?

Why do you need to know this at all? This project uses LZMA for U-Boot image by default.

ntzrmtthihu777, 2017-04-14:
ah. Didn't know if that was needed info or not.

pepe2k, 2017-04-14:
Anyway, your current questions are not related with this issue topic at all, I'm closing it.

closed 2017-04-14 by pepe2k

hanetzer, 2018-01-09:
For the sake of those googling this exact problem in the future:
The likely reason this is happening is because some package or another
has installed `/usr/include/config.h` into your root filesystem. This is a bug
in packaging; `config.h` is only meant to be used as a convenience header
at package build time, commonly used by autotools based projects, but
other projects may use `config.h` but not autotools.

The quick fix is to `rm /usr/include/config.h`, but the right fix is to use your
package manager to query which package owns this file, and submit a bug
report upstream to your distro maintainers. Google can tell you how to find
which package owns a file on your filesystem.