pepe2k/u-boot_mod #151: Please add TL-WR740N(BR) v6.0 - QCA9533

ricardojlrufino, 2017-04-21:
Hi
I would like to know what modifications I need to make to support this model:

TL-WR740N(BR) v6.0
SoC: Qualcomm Atheros QCA9533
Memory (32MB): Zentel A3S56D40GTP 
FLASH (4MB): Winbond 25Q32CSI

I made a modification and added to memory:
Windbond W25Q64 (8 MiB, JEDEC ID: EF 4017) *

But the original boot does not recognize the new size
```
Flash Manuf Id 0xef, DeviceId0 0x40, DeviceId1 0x17
flash size 4MB, sector count = 64
Flash:  4 MB
Using default environment
```

Log *before* change chip
```
U-Boot 1.1.4 (Sep 21 2015 - 16:15:07)

ap143-2.0 - Honey Bee 2.0

DRAM:  32 MB
Flash Manuf Id 0xc8, DeviceId0 0x40, DeviceId1 0x16
flash size 4MB, sector count = 64
Flash:  4 MB
Using default environment
```

ricardojlrufino, 2017-04-21:
PS: I tried **tp-link_tl-wr740n_v4**, bot not work

pepe2k, 2017-05-02:
Hi @ricardojlrufino,

> I would like to know what modifications I need to make to support this model:
> 
> TL-WR740N(BR) v6.0
> SoC: Qualcomm Atheros QCA9533
> Memory (32MB): Zentel A3S56D40GTP
> FLASH (4MB): Winbond 25Q32CSI

Platform (QCA953x) is already supported, so you don't need much to make your device supported, just numbers of GPIO-connected LEDs/buttons, default environment values, some FLASH offsets for MAC address/es, art, etc. You can follow this commit as an example: 1cd2c361697f88c5c039f4aaff5ccb3287279a00

> I made a modification and added this flash/memory:
> Windbond W25Q64 (8 MiB, JEDEC ID: EF 4017) *
> 
> But the original boot does not recognize the new size

FLASH type/size is hard-coded in original Atheros legacy U-Boot (the one which is used also by TP-Link). As soon as you don't need to access data above 4 MB limit in original U-Boot, you can use it.

> PS: I tried tp-link_tl-wr740n_v4, bot not work

It's a different platform (AR933x vs. QCA953x), you can try image for some QCA953x based TP-Link, e.g. TP-Link TL-WR841N v9.

Cheers,
Piotr

closed 2017-05-02 by pepe2k

ricardojlrufino, 2017-05-02:
I am not able to test, because of so much welding and resold, one of the flash trails was breaking ... I decided not to force more ..

Thanks for the answer.

pepe2k, 2017-05-02:
> I am not able to test, because of so much welding and resold, one of the flash trails was breaking ... I decided not to force more ..

I can recommend one of these: http://www.dediprog.com/pd

Cheers,
Piotr

ricardojlrufino, 2017-05-02:
"Update Serial Flash soldered on board with application powered or not" ...
Interesting ... Thank you.