pepe2k/u-boot_mod #152: TL-WR841N/D v11 - unknown  JEDEC ID: 0xC84016

k-plan, 2017-04-30:
Hi,

have playing around a little bit with your u-boot_mod. 
It's  neat, great work, I like it. 👍 

```
***************************************
*     U-Boot 1.1.4-aba8e6f8-clean     *
*          Build: 2017-04-25          *
***************************************

** Warning: SPI NOR FLASH in bank #1 is
   unknown, JEDEC ID: 0xC84016

   Information provided in SFDP:
   - FLASH size: 4 MB
   - erase sector size: 64 kB
   - erase sector command: 0xD8

  BOARD: TP-Link TL-WR841N/D v11
    SOC: QCA953x ver. 2 rev. 0
    CPU: MIPS 24Kc
    RAM: 32 MB DDR1 16-bit CL3-3-3-8
  FLASH: 4 MB Unknown unknown model
    MAC: 18:A6:F7:xx:xx:xx
 CLOCKS: CPU/RAM/AHB/SPI/REF
         650/400/200/ 25/ 25 MHz
```

Output from tp-link u-boot:

```
U-Boot 1.1.4 (Jun 16 2015 - 14:12:19)

ap143-2.0 - Honey Bee 2.0

DRAM:  32 MB
Flash Manuf Id 0xc8, DeviceId0 0x40, DeviceId1 0x16
flash size 4MB, sector count = 64
Flash:  4 MB
Using default environment

```

Picture of SOP8 flash chip:

[image]

Manufactor should be: GigaDevice Semiconductor

I think, it is this flash chip:

http://www.yoycart.com/Product/523722015609/

http://www.gigadevice.com/product/detail/5/119.html

The JEDEC ID: 0xC84016 I have found here:

https://lists.denx.de/pipermail/u-boot/2015-December/238635.html

Is it possible to fix `unknown  JEDEC ID`?

Thanking you in anticipation.

cu
k-plan

pepe2k, 2017-05-02:
Hi @k-plan,

> have playing around a little bit with your u-boot_mod.
> It's neat, great work, I like it. :+1:

Thanks!

> Is it possible to fix unknown JEDEC ID?

Sure, fixed in 28294e4d8844f8136e17a579b2e7340c2916d5df

Cheers,
Piotr

closed 2017-05-02 by pepe2k

k-plan, 2017-05-04:
Hi @pepe2k ,

**Confirmation:**

```

***************************************
*     U-Boot 1.1.4-28294e4d-clean     *
*          Build: 2017-05-03          *
***************************************

  BOARD: TP-Link TL-WR841N/D v11
    SOC: QCA953x ver. 2 rev. 0
    CPU: MIPS 24Kc
    RAM: 32 MB DDR1 16-bit CL3-3-3-8
  FLASH: 4 MB GigaDevice GD25Q32
    MAC: 18:A6:F7:xx:xx:xx
 CLOCKS: CPU/RAM/AHB/SPI/REF
         650/400/200/ 25/ 25 MHz

```

Works!  💯  👍  Great, thanks.

Testing press and hold reset button and web server firmware/u-boot update as well:

```
Keep button pressed for at least:
- 3s for web based recovery
- 5s for U-Boot console
- 7s for network console

. . . 0x3 seconds

HTTP server is starting for firmware update...

eth0 link down
enet1 port2 up
HTTP server is starting at IP: 192.168.1.1
HTTP server is ready!
```

Works. 😃 

You can add 4 MB GigaDevice GD25Q32 [here](https://github.com/pepe2k/u-boot_mod#supported-flash-chips) to Supported FLASH chips.

pepe2k, 2017-05-09:
Hi @k-plan,

Thanks for test and confirmation!

> You can add 4 MB GigaDevice GD25Q32 here to currently supported FLASH types.

Done in 0c183583529806349d2f80637de55f25a192789b, thanks again!

Cheers,
Piotr