pepe2k/u-boot_mod #153: why web server isn't working with ddwrt?!

Mile-Lile, 2017-05-20:
Here is my problem:

* I downloaded proper uboot you modified for my router (gl-ar150) from your server
* I renamed it to uboot.bin
* uploaded it with winscp to tmp folder
* from ssh I issued directive mtd -f write /tmp/uboot.bin uboot
* and successfully rebooted router

To check if everything went fine I typed in ssh:
strings /dev/mtd0 | grep U-Boot 

Confirmed that I upgraded u-boot to 1.1.4-0c183583-clean
[image]

But when I try to run 192.168.1.1/index.html in Chrome nothing happens (first holding reset button for 3 seconds while plugging power on). ddwrt somehow skips uboot and firmware boots... I can get only ddwrt GUI...

Strange is when I flash LEDE uboot web server works and I can upload uboot with 192.168.1.1/uboot.html with same procedure (holding reset button for 3 seconds and releasing it while powering on unit)...

Any ideas how to get uboot webserver going on ddwrt?

Mile-Lile, 2017-05-20:
sorry, you can close this one... everything is fine but it is working like this on ddwrt:
- 5s for web based recovery
- 7s for U-Boot console
- 9s for network console
everything is increased for 2 secs...
[image]

closed 2017-05-20 by Mile-Lile

pepe2k, 2017-05-21:
Hello @Mile-Lile,

> Strange is when I flash LEDE uboot web server works and I can upload uboot with 192.168.1.1/uboot.html with same procedure (holding reset button for 3 seconds and releasing it while powering on unit)...

It works like this:
- power up
- first LEDs blinks indicates network configuration, here you should push down the reset button
- while keeping reset button pressed, count LEDs blinks: min. 3 for web server, min. 5 for U-Boot CLI, min. 7 for network console

If you power up device with reset button pressed, it will start with safe clocks (400/400/200 MHz).

Cheers,
Piotr

Mile-Lile, 2017-05-21:
Thx Piotr for your answer, support and your work. I appreciate it...
I have been watching some tuts on glinet website and youtube. It works like you describe it but only with factory fw (glinet) and on LEDE... I'm trying to tell you that on DDWRT I must count 5 LEDs blink to get web server, 9 LEDs blink for netwotk console (after initial very short LED blink indicating network configuration)... didn't tested U-Boot CLI...
Cheers,
Hranislav

pepe2k, 2017-05-21:
Hi @Mile-Lile,

> Thx Piotr for your answer, support and your work. I appreciate it...

Your're welcome! :)

> I have been watching some tuts on glinet website and youtube. It works like you describe it but only with factory fw (glinet) and on LEDE... I'm trying to tell you that on DDWRT I must count 5 LEDs [...]

Firmware doesn't have any impact on how the recovery works, without changing [U-Boot environment variable `recovery`](https://github.com/pepe2k/u-boot_mod/blob/master/u-boot/include/env_scripts.h#L147) it's just impossible.

Cheers,
Piotr

Mile-Lile, 2017-05-21:
This was all new to me so I was reading your wiki to clarify these things. And that was exactly I undestand it. U-boot (with or without web server) starts before OS (firmware) from dedicated part of memory... but then I couldn't start u-boot werbserver when on ddwrt (after 3 times blinking LEDs)... so, I switched couple of times back and forward from ddwrt to lede and so on...
when switching from LEDE to DDWRT I have to use "factory flash image" that I figoured out that comes with u-boot on its own? ftp://ftp.dd-wrt.com/betas/2017/05-02-2017-r31924/GL.iNet-AR150/
web falsh without uboot? Or I am wrong?
for reverting from ddwrt to factory fw, I used https://drive.google.com/open?id=0B0SCwy1irn3qTUE1Ym1ZUjAwbUE that LOM from ddwrt made for me...

I was desperate and finally decided to update u-boot from ddwrt and did it like described in the first post.... stil couldn't bring up you web server... Then I don't know how I figured out that I have to wait 5 blinkings...
So, I can take a video and send it to you as a proof?

Mile-Lile, 2017-05-21:
OK. Here is the video https://drive.google.com/open?id=0B0SCwy1irn3qRVpvVTctOFJOV1E it 37MB so I uploaded it to my gdrive... it is not the best quality but you can see it blinks 5 times before releasing reset button...

pepe2k, 2017-05-21:
@Mile-Lile,

> This was all new to me so I was reading your wiki to clarify these things. And that was exactly I undestand it. U-boot (with or without web server) starts before OS (firmware) from dedicated part of [...]

U-Boot passes whole control to kernel, so there is no way to use code (webserver) from U-Boot in already running firmware (DD-WRT/LEDE/OpenWrt/whatever else).

> when switching from LEDE to DDWRT I have to use "factory flash image" that I figoured out that comes with u-boot on its own? ftp://ftp.dd-wrt.com/betas/2017/05-02-2017-r31924/GL.iNet-AR150/
> web falsh without uboot? Or I am wrong?

It's just firmware, you don't touch U-Boot on your device when switching between vendor, OpenWrt/LEDE and DD-WRT firmwares.

> OK. Here is the video https://drive.google.com/open?id=0B0SCwy1irn3qRVpvVTctOFJOV1E it 37MB so I uploaded it to my gdrive... it is not the best quality but you can see it blinks 5 times before releasing reset button...

Everything is correct here... first blink indicates network configuration, then you wait 4 blinks (webserver is started when number of blinks is >=3 and < 5).

Cheers,
Piotr

Mile-Lile, 2017-05-21:
ok. thx for clarifying.