pepe2k/u-boot_mod #154: after Sep 01, 2016, commit 4f11127, I can't get into webserver

r0ks, 2017-05-27:
Hi, I have a device wr703n v1, and I compile the uboot myself. Your uboot worked well before this commit. By pressing reset button for 3 seconds I can login webserver, but after commit 4f11127, I can't go into it. The router performed nothing when I press the button and power on it. If i kept pressing the button until the led on and wait, the led flashed so quick as it was in the uboot version that I can login webserver before, but it kept flashing and nothing happened, I can't login webserver.
I also had tried the version on your ftp, it didn't work too.
Can you tell me how to show the log or something else that can help you figure out what's wrong?
Thank you so much! Your uboot is so great.

pepe2k, 2017-05-27:
Hello @r0ks,

I suppose that your environment doesn't contain recovery script (it was saved before update to the version with recovery moved to environment script).

Do you have a serial console access? If yes, can you reset environment with `defenv` command and try again?

Cheers,
Piotr

r0ks, 2017-06-05:
Hi @pepe2k ,

I'm sorry that I have no serial console access, is there any way that I can set this environment script manually?

By the way my wr703n is modified to 16MB rom + 64MB ram, will this be a problem?

Thank you for your reply! 😀

pepe2k, 2017-06-08:
Hello @r0ks,

What firmware do you have on your device now?
Can you issue below command (in firmware, over SSH) and send me file `/tmp/u-boot.bin` by mail?

`cat /dev/mtd0 > /tmp/u-boot.bin`

> By the way my wr703n is modified to 16MB rom + 64MB ram, will this be a problem?

No, not at all.

Cheers,
Piotr

r0ks, 2017-06-28:
Hi @pepe2k,

I had sent the file with my google mail, please check it.
My firmware on wr703n is openwrt15.05.1.

I'm so sorry for replying late, because I was busy these days.

Thank you so much.

pepe2k, 2017-06-29:
Hi @r0ks,

> I had sent the file with my google mail, please check it.
> My firmware on wr703n is openwrt15.05.1.

I got your mail and as I was expecting, you have environment stored in FLASH and it doesn't include variable for recovery script. I'm working on a fix for that but you would need to upgrade u-boot mtd partition anyway, either using serial console access or inside OpenWrt/LEDE. If you have a serial console access, just issue `defenv` command in U-Boot CLI. If not, I highly recommend you switch to LEDE and use my [dedicated upgrade script](https://github.com/pepe2k/u-boot_mod/blob/master/target_util/u-boot-upgrade) to make it. Then you should:
- make a backup of your `/dev/mtd0` partition
- fill range `0x1ec00`~`0x1fbff` with `0xff`
- write modified data back to `/dev/mtd0` using my script

Let me know if you need any help with something here.

Cheers,
Piotr

commit bf3269ac (2017-07-09): In case of missing recovery script in env, copy it from default one

In case of update from some old version on device with correct env in
FLASH, recovery script (moved to env variable 'recovery' in 4f11127)
isn't available which results in missing 'button recovery mode'.

This fixes backward compatibility and ensures that the 'recovery'
variable is available in env - in case if it's not available in saved
env, value is copied from default env (the one embedded with image).

This fixes #154 and closes #131
closed 2017-07-09 by pepe2k