pepe2k/u-boot_mod #159: new question--how to go into the "the mode of internet" after upgard new uboot for dir505

jack338c, 2017-06-28:
hi,pepe
I can not go into the "the mode of internet" after upgard new uboot to “u-boot_mod__d-link_dir-505__20170628__git_master-2f0b6ddf.bin' by using old way--3 seconds to start the mode of internet

before I using the "2014-11-19" uboot,and everything is ok
why I upgrade uboot? because I find I can not upgrade fireware--- LEDE Reboot SNAPSHOT r4474-6371159 Kernel Version	4.4.74  for dir505
Please help me
thanks
Jack

closed 2017-06-28 by jack338c

pepe2k, 2017-06-29:
Hi @jack338c,

Did you solve your problem?

I just upgraded my DIR-505 rev. A1 with [u-boot_mod__d-link_dir-505__20170628__git_master-2f0b6ddf.bin](http://projects.dymacz.pl/u-boot_mod/u-boot_mod__d-link_dir-505__20170628__git_master-2f0b6ddf.bin) and I don't have any problems with web recovery mode - just make sure to use the `WPS` button, not the `reset` one (DIR-505 differs here from other supported devices but for backward compatibility I will keep it as it is now).

```
***************************************
*     U-Boot 1.1.4-2f0b6ddf-clean     *
*          Build: 2017-06-28          *
***************************************

** Warning: bad env CRC, using default,
   use 'saveenv' to save it in FLASH

  BOARD: D-Link DIR-505
    SOC: AR9330 rev. 1
    CPU: MIPS 24Kc
    RAM: 64 MB DDR2 16-bit CL3-4-4-10
  FLASH: 8 MB Winbond W25Q64
    MAC: 00:03:7F:09:0B:AD (fixed)
 CLOCKS: CPU/RAM/AHB/SPI/REF
         400/400/200/ 20/ 25 MHz

Keep button pressed for at least:
- 3s for web based recovery
- 5s for U-Boot console
- 7s for network console

. . . 0x3 seconds

HTTP server is starting for firmware update...

Ethernet mode (duplex/speed): 1/100 Mbps
HTTP server is starting at IP: 192.168.1.1
HTTP server is ready!
```

I also upgraded it to LEDE snapshot image and everything works just fine.

If it still doesn't work for you, could please show me output from below command in LEDE:
`hexdump /dev/mtd2`

Cheers,
Piotr

jack338c, 2017-06-30:
It works fine ! Awesome !Thank you!