pepe2k/u-boot_mod #160: Ubiquiti Rocket M 2 support

efidrin, 2017-07-06:
Dear pepe2K,  is support for UBNT-RM planned ?
It have AR9344 and 128 RAM. I have this device and can public dumps and research leds.
Or may be, I will try to make config (u-boot/Makefile) myself ?

pepe2k, 2017-07-09:
Hi @efidrin,

> Dear pepe2K, is support for UBNT-RM planned ?

I (in most cases) add support for devices I have access to or I know well (plus, I also accept pull requests from people who own the hardware). The one you are asking about I don't know at all (and in fact, I wasn't able to find any useful information about it), so the answer would be 'no, it's not planned'.

> It have AR9344 and 128 RAM. I have this device and can public dumps and research leds.

That could be not enough. Is this device supported in OpenWrt/LEDE? Can you share full name of the device (`UBNT-RM` doesn't tell me and uncle Google much).

> Or may be, I will try to make config (u-boot/Makefile) myself ?

Pull requests are always welcome! :)

Cheers,
Piotr

efidrin, 2017-07-10:
>That could be not enough. Is this device supported in OpenWrt/LEDE? Can you share full name of the device (UBNT-RM doesn't tell me and uncle Google much).

Device info [in pdf](https://dl.ubnt.com/datasheets/rocketmgps/RocketM_Ti_DS.pdf)

I used OpenWrt (openwrt-ar71xx-generic-ubnt-rocket-m-ti-squashfs-factory) and LEDE (lede-17.01.2-ar71xx-generic-ubnt-rocket-m-ti-squashfs-sysupgrade). Prefix -ti- aded in 15.05.1. 
With OpenWrt I had some problems witch have all [Airmax famili](https://wiki.openwrt.org/toh/ubiquiti/airmaxm). Even downgrading firmware helped, only with fwupdate option for flashing u-boot (sorry but can`t remember or find, what exactly option it was).

If it necessary , i can share boot log with u-boot part.

Thu Jul  6 23:55:16 2017 kern.notice kernel: [    0.000000] Linux version 4.4.71 (buildbot@builds-02.infra.lede-project.org) (gcc version 5.4.0 (LEDE GCC 5.4.0 r3101-bce140e) ) #0 Wed Jun 7 19:24:41 2017
Thu Jul  6 23:55:16 2017 kern.debug kernel: [    0.000000] MyLoader: sysp=ffe58f7f, boardp=7f8e5b1c, parts=6ea75b7a
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] CPU0 revision is: 0001974c (MIPS 74Kc)
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] SoC: Atheros AR9344 rev 2
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] Determined physical RAM map:
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000]  memory: 08000000 @ 00000000 (usable)
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Thu Jul  6 23:55:16 2017 kern.warn kernel: [    0.000000] No valid device tree found, continuing without
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] Zone ranges:
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] Movable zone start for each node
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] Early memory node ranges
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Thu Jul  6 23:55:16 2017 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Thu Jul  6 23:55:16 2017 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 803d04d0, node_mem_map 81000000
Thu Jul  6 23:55:16 2017 kern.debug kernel: [    0.000000]   Normal zone: 256 pages used for memmap
Thu Jul  6 23:55:16 2017 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Thu Jul  6 23:55:16 2017 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Thu Jul  6 23:55:16 2017 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Thu Jul  6 23:55:16 2017 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Thu Jul  6 23:55:16 2017 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Thu Jul  6 23:55:16 2017 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
Thu Jul  6 23:55:16 2017 kern.notice kernel: [    0.000000] Kernel command line:  board=UBNT-RM-TI mtdparts=spi0.0:256k(u-boot)ro,64k(u-boot-env)ro,7552k(firmware),256k(cfg)ro,64k(EEPROM)ro console=ttyS0,115200 rootfstype=squashfs,jffs2 noinitrd
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] Writing ErrCtl register=00000000
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] Readback ErrCtl register=00000000
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] Memory: 125328K/131072K available (3073K kernel code, 163K rwdata, 412K rodata, 312K init, 205K bss, 5744K reserved, 0K cma-reserved)
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] NR_IRQS:51
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] Clocks: CPU:560.000MHz, DDR:450.000MHz, AHB:225.000MHz, Ref:40.000MHz
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6825930166 ns
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.000009] sched_clock: 32 bits at 280MHz, resolution 3ns, wraps every 7669584382ns
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.008298] Calibrating delay loop... 278.93 BogoMIPS (lpj=1394688)
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.081137] pid_max: default: 32768 minimum: 301
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.086171] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.093221] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[... cut ...]
Thu Jul  6 23:55:16 2017 kern.debug kernel: [    0.634497] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
Thu Jul  6 23:55:16 2017 kern.debug kernel: [    0.634731] pci_bus 0000:00: busn_res: [bus 00-ff] end is updated to 00
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.634766] pci 0000:00:00.0: BAR 0: assigned [mem 0x10000000-0x1000ffff 64bit]
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.642563] pci 0000:00:00.0: using irq 40 for pin 1
Thu Jul  6 23:55:16 2017 kern.info kernel: [    0.648584] clocksource: Switched to clocksource MIPS

pepe2k, 2017-07-10:
Hi @efidrin,

As I wrote before, if you want this devices to be supported, it's up to you to prepare, test required changes and eventually send PR.

Cheers,
Piotr

closed 2017-07-10 by pepe2k