pepe2k/u-boot_mod #164: Error: failed to execute 'bootcmd'! and warnings during compilation

Tarask1n, 2017-07-23:
branch: master
build: a33dd2934f158c0cb069758c9a296338b3cef560
toolchain: OpenWrt-Toolchain-ar71xx-generic_gcc-5.3.0_musl-1.1.16.Linux-x86_64.tar.bz2

Load output:
`
***************************************
*     U-Boot 1.1.4-a33dd293-dirty     *
*          Build: 2017-07-23          *
***************************************

** Warning: bad env CRC, using default,
   use 'saveenv' to save it in FLASH

  BOARD: TP-Link TL-WDR43x0
    SOC: AR9344 rev. 2
    CPU: MIPS 74Kc
    RAM: 128 MB DDR2 32-bit CL3-4-4-10
  FLASH: 16 MB Winbond W25Q128
   PCIe: 168C:0030
    MAC: 00:03:7F:09:0B:AD (fixed)
 CLOCKS: CPU/RAM/AHB/SPI/REF
         550/400/200/ 25/ 40 MHz

Hit any key to stop booting:  0

Booting image from 0x9F020000...
## Error: unsupported image header

## Error: failed to execute 'bootcmd'!
   Starting web server for update...
`

Compilation warnings:
`
taras@taras-ubuntu:~/uboot/u-boot_mod$ make clean_all
Making clean...
Making clobber...
Removing configuration...
Removing all binary images...
taras@taras-ubuntu:~/uboot/u-boot_mod$ make tp-link_tl-wdr43x0
Removing configuration...
Preparing configuration for target: tp-link_tl-wdr43x0

  Device vendor/model:	TP-Link TL-WDR43x0
  Custom recovery web:	no
  Image size limit:	123 KB
  Default FLASH size:	8 MB
  GPIO reset button:	16
  Button active low:	yes
  LZMA compression:	yes
  RAM-loadable only:	no

Preparing web server files...

a - crc32.o
a - ctype.o
a - display_options.o
a - string.o
a - LzmaDecode.o
a - LzmaWrapper.o
a - vsprintf.o
a - main.o
a - cmd_bootm.o
a - cmd_boot.o
a - cmd_flash.o
a - cmd_mem.o
a - cmd_custom.o
a - cmd_net.o
a - cmd_nvedit.o
a - cmd_itest.o
a - cmd_load.o
a - cmd_qcaclk.o
a - command.o
a - console.o
a - devices.o
a - dlmalloc.o
a - environment.o
a - env_common.o
a - env_nowhere.o
a - exports.o
a - flash.o
a - lists.o
a - env_flash.o
a - hush.o
a - xyzModem.o
a - crc16.o
a - s_record.o
a - board.o
a - time.o
a - mips_linux.o
a - netconsole.o
a - net.o
a - tftp.o
a - rarp.o
a - eth.o
a - bootp.o
a - sntp.o
a - httpd.o
a - date.o
a - db12x.o
a - ../common/spi_flash.o
a - ../common/common.o
a - ../common/athrs17_phy.o
a - uip.o
a - uip_arch.o
a - uip_arp.o
a - httpd.o
a - fs.o
In file included from /home/taras/uboot/u-boot_mod/u-boot/include/asm/io.h:18:0,
                 from ag934x.c:6:
ag934x.c: In function 'ag7240_mii_setup':
ag934x.c:244:29: warning: array subscript is above array bounds [-Warray-bounds]
    ag7240_reg_wr(ag7240_macs[1], AG7240_MAC_MII_MGMT_CFG, mgmt_cfg_val | (1 << 31));
                             ^
/home/taras/uboot/u-boot_mod/u-boot/include/asm/addrspace.h:54:57: note: in definition of macro 'KSEG1ADDR'
 #define KSEG1ADDR(a)  ((__typeof__(a))(((unsigned long)(a) & 0x1fffffff) | KSEG1))
                                                         ^
/home/taras/uboot/u-boot_mod/u-boot/include/ar7240_soc.h:542:42: note: in expansion of macro 'ar7240_reg_wr_nf'
 #define ar7240_reg_wr(_phys, _val)  do { ar7240_reg_wr_nf(_phys, _val); ar7240_reg_rd(_phys); } while(0);
                                          ^
ag934x.h:38:39: note: in expansion of macro 'ar7240_reg_wr'
 #define ag7240_reg_wr(_mac, _x, _y)   ar7240_reg_wr(((_x) + _mac->mac_base), (_y))
                                       ^
ag934x.c:244:4: note: in expansion of macro 'ag7240_reg_wr'
    ag7240_reg_wr(ag7240_macs[1], AG7240_MAC_MII_MGMT_CFG, mgmt_cfg_val | (1 << 31));
    ^
ag934x.c: In function 'ag7240_miiphy_read':
ag934x.c:19:43: warning: array subscript is above array bounds [-Warray-bounds]
 #define ag7240_unit2mac(_unit) ag7240_macs[(_unit)]
                                           ^
ag934x.c:20:53: note: in expansion of macro 'ag7240_unit2mac'
 #define ag7240_name2mac(name) strcmp(name,"eth0") ? ag7240_unit2mac(1) : ag7240_unit2mac(0)
                                                     ^
ag934x.c:669:22: note: in expansion of macro 'ag7240_name2mac'
  ag7240_mac_t *mac = ag7240_name2mac(devname);
                      ^
ag934x.c: In function 'ag7240_miiphy_write':
ag934x.c:19:43: warning: array subscript is above array bounds [-Warray-bounds]
 #define ag7240_unit2mac(_unit) ag7240_macs[(_unit)]
                                           ^
ag934x.c:20:53: note: in expansion of macro 'ag7240_unit2mac'
 #define ag7240_name2mac(name) strcmp(name,"eth0") ? ag7240_unit2mac(1) : ag7240_unit2mac(0)
                                                     ^
ag934x.c:706:22: note: in expansion of macro 'ag7240_name2mac'
  ag7240_mac_t *mac = ag7240_name2mac(devname);
                      ^
a - qca_common.o
a - qca_clocks.o
a - qca_sf.o
a - qca_dram.o
a - qca_pci.o
a - qca_ls_uart.o
a - ag934x.o
a - qca_gpio_init.o
a - qca95xx_pll_init.o
a - cpu.o
a - cache.o
a - bootstrap_board.o
a - LzmaDecode.o
a - string.o
a - LzmaWrapper.o

Building lzma host utility...

Compressing U-Boot image lzma_host...
Preparing LZMA compressed U-Boot image u-boot.lzimg...
    Image Name: U-Boot 1.1.4-a33dd293-dirty
       Created: Sun Jul 23 09:21:52 2017
    Image Type: MIPS Linux Firmware (lzma compressed)
     Data Size: 56699 Bytes = 55.37 kB = 0.05 MB
  Load Address: 0x80010000
   Entry Point: 0x80010000

Merging bootstrap.bin with u-boot.lzimg...
    Bootstrap size: 12492 Bytes
   LZMA image size: 56763 Bytes
  Total image size: 69255 Bytes

Copying compiled image...
Checking size of the image...
Preparing 123 KB image padded with 0xFF...
Preparing final image...
Calculating MD5 sum for the final image...

DONE!
Image 'bin/u-boot_mod__tp-link_tl-wdr43x0__20170723__git_master-a33dd293.bin' is ready!
`

pepe2k, 2017-07-23:
Hello @Tarask1n,

> `Booting image from 0x9F020000...`
> `## Error: unsupported image header`

Your device doesn't have supported image at offset 0x9F020000 in FLASH. Also, your MAC address is missing in FLASH (I suppose you didn't make copy of first 128 KB on FLASH before writing modified U-Boot):

> `MAC: 00:03:7F:09:0B:AD (fixed)`

> Compilation warnings:

Compilation warnings are just warnings, you can ignore them, they are not related anyway.

Cheers,
Piotr

closed 2017-07-23 by pepe2k

Tarask1n, 2017-07-23:
Hi Piotr,

Please, put your attention on that issue, because when I compile 59d12316 build it enters to bootcmd perfectly AT THE SAME ENVIRONMENT (without MAC, image and the same warnings) and console is accessible. Any changes from my side to code, just checkout on 59d12316 and remake.

```
***************************************
*     U-Boot 1.1.4-59d12316-dirty     *
*          Build: 2017-07-23          *
***************************************

** Warning: bad env CRC, using default,
   use 'saveenv' to save it in FLASH

  BOARD: TP-Link TL-WDR43x0
    SOC: AR9344 rev. 2
    CPU: MIPS 74Kc
    RAM: 128 MB DDR2 32-bit CL3-4-4-10
  FLASH: 16 MB Winbond W25Q128
   PCIe: 168C:0030
    MAC: 00:03:7F:09:0B:AD (fixed)
 CLOCKS: CPU/RAM/AHB/SPI/REF
         550/400/200/ 25/ 40 MHz

Hit any key to stop booting:  0

Booting image from 0x9F020000...
## Error: unsupported image header
u-boot>
```

pepe2k, 2017-07-23:
Hi @Tarask1n,

TL-WDR4300, current snapshot image:

```
***************************************
*     U-Boot 1.1.4-a33dd293-clean     *
*          Build: 2017-07-12          *
***************************************

** Warning: bad env CRC, using default,
   use 'saveenv' to save it in FLASH

  BOARD: TP-Link TL-WDR43x0
    SOC: AR9344 rev. 2
    CPU: MIPS 74Kc
    RAM: 128 MB DDR2 32-bit CL3-4-4-10
  FLASH: 8 MB Winbond W25Q64
   PCIe: 168C:0033
    MAC: 10:FE:FA:xx:xx:xx
 CLOCKS: CPU/RAM/AHB/SPI/REF
         550/400/200/ 25/ 40 MHz

Hit any key to stop booting:  0

Booting image from 0x9F020000...

   Vendor/image name:    OpenWrt r49007
   Hardware ID:          0x43000001
   Whole image size:     7.8 MB (8126464 bytes)
   Kernel size:          1.1 MB (1181184 bytes)
   Rootfs size:          5.8 MB (6117744 bytes)
   Kernel load address:  0x80060000
   Kernel entry point:   0x80060000

   Header CRC...  skipped
   Data CRC...    skipped

Stopping network... OK!
Uncompressing Kernel... OK!
Starting kernel...

[    0.000000] Linux version 3.18.27 (cezary@eko.one.pl) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 unknown) ) #7 Sat Mar 19 06:08:30 CET 2016
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001974c (MIPS 74Kc)
[    0.000000] SoC: Atheros AR9344 rev 2
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x07ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x07ffffff]
[    0.000000] Initmem setup node 0 [mem 0x00000000-0x07ffffff]
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line:  board=TL-WDR4300  console=ttyS0,115200 rootfstype=squashfs,jffs2 noini
```
Show me output from the following commands (in U-Boot): `printenv`, `imi 0x9F020000`, `md.l 0x9F020000`.

Cheers,
Piotr

Tarask1n, 2017-07-23:
Hi Piotr,

With FW all goes Ok. 
Please, try to enter boot console by pressing Esc due boot or erase FW and uboot shows you:
```
## Error: failed to execute 'bootcmd'!
   Starting web server for update...
```
and it doesn't start console and switch to web restore.

pepe2k, 2017-08-01:
Hi @Tarask1n,

> [...]
> and it doesn't start console and switch to web restore.

Yes, this is expected: https://github.com/pepe2k/u-boot_mod/commit/e5c09f2374bb758d4187db43ecd8c8387bcacc05

Cheers,
Piotr

Tarask1n, 2017-08-02:
Hi Piotr,

I saw you commit. But may be I missed something. In my opinion it is illogical when I see "Hit any key to stop booting:" message then I press Esc key and expecting to be in boot console, but I moved to FW/boot restore procedure.

Best regards,
Taras.

pepe2k, 2017-08-03:
Hi @Tarask1n,

> I saw you commit. But may be I missed something. In my opinion it is illogical when I see "Hit any key to stop booting:" message then I press Esc key and expecting to be in boot console, but I moved to FW/boot restore procedure.

I'm not able to reproduce this problem... could you please show me your changes in code?

Cheers,
Piotr