pepe2k/u-boot_mod #170: uboot with AR9331 strange problem

skyfirecctv, 2017-08-08:
Hello pep2k
we are working with AR9331 as WIFI router/client, the DDR2 we used is winbond W9751g6kb_a09
during the QA testing we found a strange problem, if AR9331 working as AP mode, some of the PCBA will reboot without any log (reboot caused by hardware watchdog inside AR9331) after powering up 10-30mins.
if AR9331 working as station mode, will getting worse, 5 out of 10 PCBA will rebooting after power up 10-30mins, and some of the PCBA will keep rebooting.
Also we tried to unload the wifi driver but still no luck
We have tried using original Atheros Uboot and the uboot you modified, but nothing changes,
Our PLL working as following
CPU 400//DDR2 400//AHB 200

Also we found this commit 
https://github.com/pepe2k/u-boot_mod/commit/aba8e6f81d6b84592a487526bddc33d916940bba
seems the exactly problem we faced 
but after upgrading to that version nothing changes…
so do you have any clue for this problem? 
Thank you so much!

K

pepe2k, 2017-08-08:
Hello @skyfirecctv,

Have a look at discussion in #120 and let me know if that change solves your problems.

I still wasn't able to reproduce this issue on any AR9331+DDR2 platform I have access to and decided to not merge it. Could you also tell me something more about your device, is it a custom design PCB or some ready module/router?

Cheers,
Piotr

skyfirecctv, 2017-08-08:
hi @pepe2k it is a custom design PCB but very much like the AP121 reference design, i will try #120 today and let you know the results
Cheers
K

skyfirecctv, 2017-08-09:
hi @pepe2k just one more update,
originally AP121 design is using DDRI so our hardware engineer take the AR9342's DDR2 ref design
[image]
as you can see the DQ0-7 and DQ8-15 pin are 'swaped' with DATA lines
also LDQS is connected with DQS1 UDQS is connected with DQS0

but after long searching on the internet we found some ref. design which uses DDR2 with AR9331
the schematic looks like

as you can see all the DQ pins are connected with DATA lines in order, and the pin LDQS is connected with DQS0 and UDQS is connected with DQS1

[image]

any clue with that?
thanks!

pepe2k, 2017-08-09:
Hello @skyfirecctv,

I don't have access to QC/A reference designs but I suppose if the DDR chip wasn't connected correctly, your board wouldn't boot at all/ U-Boot relocates itself from FLASH to upper area in RAM (highest address) and then continue execution from there so if your board boots, different RAM chip connection isn't the case here.

I'm almost sure that #120 will solve your problems.

Cheers,
Piotr

skyfirecctv, 2017-08-11:
hi @pepe2k we have tried the following changes
[image]
but still rebooting.....
nothing changed

B.R.
K

pepe2k, 2017-08-11:
Hi @skyfirecctv,

Look again at the patch, there are two changes in code, not just one:
[image]

Cheers,
Piotr

skyfirecctv, 2017-08-11:
hi @pepe2k  I confirmed with the software dev guy , they did both of the changes
but still with the same results
B.R.
K

pepe2k, 2017-08-11:
Hi @skyfirecctv,

> We have tried using original Atheros Uboot and the uboot you modified, but nothing changes,

Does this mean that your problem is visible also with Atheros U-Boot version? If yes, I would say then something is wrong with your hardware design. I have few different AR9331 + DDR2 boards and all of them are working just fine.

What you can eventually do is to lower the clocks (maybe try 300/300/150 MHz), just to check if it has any impact on stability and/or manually change CAS latency with `CONFIG_BOARD_DRAM_CAS_LATENCY`.

Cheers,
Piotr

closed 2017-12-08 by pepe2k

raszpl, 2017-12-09:
@skyfirecctv 
>as you can see the DQ0-7 and DQ8-15 pin are 'swaped' with DATA lines

doesnt matter, as long as all are length matched and not crossing on top of each other

landswipe, 2019-05-15:
@skyfirecctv Can you post direct links to the DDR schematics you found? Also, can you post the part number of the DRAM memory part you are actually using in your designs?