pepe2k/u-boot_mod #171: U-Boot console

jack338c, 2017-08-09:
Dear Pepe
Could you please teach me how to get into U-Boot console,I want change mac for new flash

Curruntly,I'm use your u-boot_mod__tp-link_tl-wr802n__20170712__git_master-a33dd293
I can use  www mode.
I was try to read your blog,but it's hard to understand the part of U-Boot console by translation
I also was try to use nc -u -p 6666 192.168.1.1 6666 on my ubuntu,but it didnot work yet
please help me

by the way I have areadly flahs lede firmware to my 802n,but can now do a factory reset by hold reset button

pepe2k, 2017-08-09:
Hi @jack338c!

I suppose you are asking about **network** console (in other words: `netconsole`)? You can start `netconsole` in two ways:

- manually, inside U-Boot CLI with `startnc` command, or
- using reset button: keep it pressed for at least 7 blinks after first one (which indicates network configuration)

After that, you can connect to the router under any Linux system with `nc -u -p 6666 192.168.1.1 6666`. Just make sure your PC has a static IP 192.168.1.2/24 and your firewall is disabled!

Cheers,
Piotr

closed 2017-08-09 by pepe2k

jack338c, 2017-08-09:
yes, use  network console

thank you very much for your help
 I will try it

jack338c, 2017-08-10:
@pepe2k 
I tryed it by nc -u -p 6666 192.168.1.1 6666 ,but it's unsuccessful too
here are my steps
1.conect router and ubuntu pc with network cable and set  a static IP: 192.168.1.2 24(netmask) 192.168.1.1(gatway)
2.release reset button after 7 blinks 
3.open terminal and enter nc -u -p 6666 192.168.1.1 6666 ,but no response

by the way :I can use reset button to do a factory reset for dir505,but why my tplink wr802n can not 

thanks

pepe2k, 2017-08-10:
Hi @jack338c!

> 4.open terminal and enter nc -u -p 6666 192.168.1.1 6666 ,but no response

Did you try to send a command after this? Did you disable firewall on your PC? You can check if there is any communication over Ethernet using e.g. Wireshark or tcpdump.

Cheers,
Piotr

jack338c, 2017-08-10:
@pepe2k 

Yes I do. I will try again later .thanks for your patience and help.

Did you try to send a command after this?(setp 4) Did you disable firewall on your PC?(step 2)You can check if there is any communication over Ethernet using e.g. Wireshark or tcpdump.

jack338c, 2017-08-18:
here are my steps
1.conect router and ubuntu pc with network cable and set a static IP: 192.168.1.2 24(netmask) 
192.168.1.1(gatway) on pc

2.disable fireware by sudo ufw disable ( sudo ufw status :inactive)

3.release reset button after 7 blinks

4.open terminal and enter" nc -u -p 6666 192.168.1.1 6666 " but no response

jack338c, 2017-08-18:
Deer pepe

I can't do this yet,please help check my steps ,is that right?
thank you

steps
1.use network cable conect router and ubuntu pc and set a static IP: 192.168.1.2 24(netmask) 
192.168.1.1(gatway) on pc

2.disable fireware by sudo ufw disable ( sudo ufw status :inactive)

3.release reset button after 7 blinks

4.open terminal and enter" nc -u -p 6666 192.168.1.1 6666 " but no response

and here is some log in terminal

jack@jack-ThinkPad-T400:~$ sudo ufw status
[sudo] password for jack: 
Status: inactive
jack@jack-ThinkPad-T400:~$ nc -u -p 6666 192.168.1.1 6666
^Z
[1]+  Stopped                 nc -u -p 6666 192.168.1.1 6666
jack@jack-ThinkPad-T400:~$ nc -u -p 6666 192.168.1.1 6666
nc: bind failed: Address already in use
jack@jack-ThinkPad-T400:~$

noeldum, 2018-09-28:
Same problem here using WR703N. 
Using -v for nc shows there is a connection.
However no response from the device at all is shown. No prompt.
Don't have easy access to a serial interface to do the startnc.

jack338c, 2019-05-06:
@pepe2k 
Dear pepe2k
I have successfully connected u-boot network console via ubuntu 18.04

but when I sent commond "setmac" ,it shows following words,Could you tell what 's the commond for change mac ,like change mac of wifi or lan,thanks

 uboot> setmac
setmac
## Error: unknown command 'setmac' - try 'help'
uboot>

jack338c, 2019-05-06:
my bad ,version is different,I solved it
thanks