pepe2k/u-boot_mod #172: Netgear R6100 Support

lgrootnoob, 2017-08-12:
This is an ideal device to support https://wiki.openwrt.org/toh/hwdata/netgear/netgear_r6100
it comes with uboot 1.1.4. https://wiki.openwrt.org/toh/netgear/netgear_r6100#openwrt_bootlog
If you are interested and would like the img please let me know.

lgrootnoob, 2017-08-12:
Source code is available on netgear's site here https://kb.netgear.com/2649/NETGEAR-Open-Source-Code-for-Programmers-GPL

pepe2k, 2017-08-13:
Hello @lgrootnoob,

Please, see my answer for a similar question here: #168 (comment)

Cheers,
Piotr

closed 2017-08-13 by pepe2k

lgrootnoob, 2017-08-13:
Thats fine, if you can explain the porting process and outline the directories with device specific bits. Maybe show me a couple of commits that were used to add support for another device so I can get started.