pepe2k/u-boot_mod #180: Winbond 25Q256 chip support

kenzhoanandiya, 2017-12-07:
Hi @pepe2k 

I want to mod my router WDR4300 with winbond 25Q256FVFG 32Mb flash rom to expand my internal storage device, using openwrt image. it's that possible to do it on this uboot? how to re partition the flash rom?

pepe2k, 2017-12-08:
Hi @kenzhoanandiya

> I want to mod my router WDR4300 with winbond 25Q256FVFG 32Mb flash rom to expand my internal storage device, using openwrt image.

Are you aware of that this would not work out of the box? You would need to perform a small hardware modification (connect reset lines between SOC and SPI NOR chip) and prepare custom OpenWrt/LEDE image because ART data in mach-*.c file is accessed using memory mapped SPI (which is limited to 16 MB range).

> it's that possible to do it on this uboot?

Not yet but actually... it will be possible to use >16 MB in my modification soon, it's already WIP.

> how to re partition the flash rom?

You don't have to do it, for this TP-Link OpenWrt/LEDE uses dedicated mtd splitter which will adjust partition sizes/offsets. But as I wrote above, it's not a simple chip swap as with 16 MiB ones.

Cheers,
Piotr

closed 2017-12-08 by pepe2k

lePereT, 2020-08-01:
Hi @pepe2k just wondered what the situation is currently with 32MB flash chips? I'm hoping to do a swap on a device tree based Ath79 device (a QCA9531 device), does that change things regarding the "ART data in mach-*.c file" that you mentioned?

Would love to know the steps now needed for larger flash, and then maybe I can try making some progress.