pepe2k/u-boot_mod #182: Web failsafe httpd question

zhangsd20, 2017-12-08:
Hi pepe2k 
    HTTP server is starting at IP: 192.168.1.1 
    HTTP server is ready!
but my pc can't open the web and the ping failed ! why? 
Thanks !

pepe2k, 2017-12-08:
Hi @zhangsd20

Is your PC configured with a static IP 192.168.1.x/24 (different than 192.168.1.1)?

Cheers,
Piotr

zhangsd20, 2017-12-11:
Good morning：
            My pc IP 192.168.1.10 ，but  the problem is not there，I have found the problem and solved it。
Thank you very mach！

mh4nx7net, 2021-02-22:
**no page found -- help**

[image]

```
    RAM: 32 MB DDR1 16-bit CL3-3-3-8
  FLASH: 4 MB Winbond W25Q32
    MAC: 00:03:7F:09:0B:AD
 CLOCKS: CPU/RAM/AHB/SPI/REF
         550/400/200/ 25/ 25 MHz

Keep button pressed for at least:
- 3s for web based recovery
- 5s for U-Boot console
- 7s for network console

. . . . 0x4 seconds

HTTP server is starting for firmware update...

Ethernet mode (duplex/speed): 1/100 Mbps
HTTP server is starting at IP: 192.168.1.1
HTTP server is ready!

Request for: /cgi-bin/luci/
## Error: file not found!
Request for: /cgi-bin/luci/style.css
## Error: file not found!
Request for: /favicon.ico
## Error: file not found!
Request for: /cgi-bin/luci/
## Error: file not found!
Request for: /cgi-bin/luci/style.css
## Error: file not found!
Request for: /favicon.ico
## Error: file not found!
Request for: /cgi-bin/luci/
## Error: file not found!
Request for: /cgi-bin/luci/style.css
## Error: file not found!
Request for: /favicon.ico
## Error: file not found!
```

i cant found any index.html by this repo :( where i can found them ?

mpratt14, 2021-02-22:
http://192.168.1.1/index.html

closed 2022-10-13 by pepe2k