pepe2k/u-boot_mod #183: wr703n art image

jekkos, 2017-12-25:
Hi there
Some great work you did here. I upgraded my wr703n with a bigger flash chip and this bootloader. In the process I lost the original flash without having a backup (was bricked anyway).
Now I recovered an art partition from another wr703n board.
The wifi driver seems to load correctly although it has no mac address (all FF)
Any idea how this works? Seems that I can't get the interface created for some reason, although it does seem to initialize. Any idea if this might work at all? I saw you wrote some tooling for modifying art images for different TPlink routers. Thanks!

```
[   10.206678] ath: EEPROM regdomain: 0x0
[   10.206708] ath: EEPROM indicates default country code should be used
[   10.206722] ath: doing EEPROM country->regdmn map search
[   10.206747] ath: country maps to regdmn code: 0x3a
[   10.206763] ath: Country alpha2 being used: US
[   10.206777] ath: Regpair used: 0x3a
[   10.218534] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   10.223382] ieee80211 phy0: Atheros AR9330 Rev:1 mem=0xb8100000, irq=2
```

noeldum, 2019-02-05:
What abut the setmac command from u-boot prompt?

jekkos, 2019-02-05:
Ok cool didn't know about that one! What I did in the end was a hexedit of the bootloader here and added the mac adress in the correct place (uboot-env section). After that wifi came up correctly and so victory. Router is still serving as a repeater in my mesh network now.

closed 2019-02-05 by jekkos