pepe2k/u-boot_mod #184: Support for WNDR3700v4

GXTX, 2017-12-31:
I'd love to be able to have a bit more control over the bootloader in this little guy.
https://wiki.openwrt.org/toh/hwdata/netgear/netgear_wndr3700_v4

GXTX, 2018-01-01:
`U-Boot 1.1.4 (Jun 21 2012 - 11:09:33)

DNI HW ID: 29763948 flash 128MB RAM 128MB U-boot dni29 V1.8

DRAM:  128 MB
Atheros on-chip NAND FLash Controller Driver, Version 0.1 (c) 2010 Atheros Communications, Ltd.
Ath Nand ID[87ff0178]: 2c:f1:80:95:02
ONFI MICRON      MT29F1G08ABADAWP   
Micron NAND 128MiB 3,3V 8-bit [128MB]
====== NAND Parameters ======
sc = 0x87ff0158 bbt = 0x87f68008 bbt_size = 0x100 nf_ctrl = 0x344
page = 0x800 block = 0x20000 oob = 0x40
size = 128MB
Setting 0xb8116290 to 0x61c22d0f
Using default environment

In:    serial
Out:   serial
Err:   serial
Net:   ag934x_enet_initialize...
 4096 bytes read: OK
Fetching MAC Address from 0x87fe7928
 wasp  reset mask:c02300 
WASP  ----> S17 PHY *
: cfg1 0x80000000 cfg2 0x7114
eth0: c4:04:15:91:03:a0
athrs17_reg_init: complete
eth0 up
eth0
Hit any key to stop autoboot:  0

ar7240> printenv
bootargs=console=ttyS0,115200 root=31:03 rootfstype=jffs2 init=/sbin/init mtdparts=ath-nand:128k(u-boot),384k(u-boot-env),1280k(uImage),7m(rootfs),128k(dummy),128k(caldata)
bootcmd=nmrp;if loadn_dniimg 0 0x6c0000 0x81000000 && chk_dniimg 0x81000000; then bootm 0x81000000;else fw_recovery; fi
baudrate=115200
ethaddr=0x00:0xaa:0xbb:0xcc:0xdd:0xee
ipaddr=192.168.1.1
serverip=192.168.1.7
dir=
lu=tftp 0x80060000 ${dir}2fw.bin&&nand erase 0x0 0x20000&&nand write $fileaddr 0x0 $filesize
lf=tftp 0x80060000 ${dir}db12x${bc}-nand-jffs2${ns}&&nand erase 0x1c0000 0x700000&&nand write $fileaddr 0x1c0000 $filesize
lk=tftp 0x80060000 ${dir}vmlinux${bc}.lzma.uImage&&nand erase 0x80000 0x140000&&nand write $fileaddr 0x80000 $filesize
stdin=serial
stdout=serial
stderr=serial
bootdelay=2
ethact=eth0

Environment size: 792/262140 bytes
ar7240> `

pepe2k, 2018-01-01:
Hello @GXTX,

Please, see my answer for a similar question here: https://github.com/pepe2k/u-boot_mod/issues/168#issuecomment-319584899

Cheers,
Piotr

closed 2018-01-01 by pepe2k