pepe2k/u-boot_mod #187: Upgrade WR703N u-boot_mod from 64kB to 123kB version

radiolee, 2018-02-05:
Hi, I was flashed 20130422 u-boot mod before. Today I download the u-boot_mod__tp-link_tl-wr703n_v1__20180204__git_master-4fbf43bf.bin pre-build and get problem when upgrade it in websafe http://192.168.1.1/uboot.html, the warnings says "you should upload a file with size of exactly 64 KB", so I am not upload it.
I can't find the document about 64kb to 123kb except this committed :
https://github.com/pepe2k/u-boot_mod/commit/7829f50c0e92024fde613cb01e65cbdeae1f126b
So I am not sure is it safe to upgrade the u-boot in websafe, or I must update it via tftp under serial/net console?

pepe2k, 2018-02-05:
Hi @radiolee,

> So I am not sure is it safe to upgrade the u-boot in websafe, or I must update it via tftp under serial/net console?

Please, update using dedicated script (it requires up-to-date LEDE/OpenWrt image installed on device): https://github.com/pepe2k/u-boot_mod/blob/master/target_util/u-boot-upgrade. You can use serial/tftp but you must remember to combine new image with existing data on flash in second 64 KB sector.

Cheers,
Piotr

radiolee, 2018-02-05:
thanks very much

closed 2018-02-05 by radiolee